CadQuery/cadquery #657: Is it possible to list/modify parameters of imported model?

welenely, 2021-02-24:
Hello, I am pretty new to CAD stuff and the question is: Is there a way to list/modify parameters of imported model? 
The desired algorythm is next:
1. Model is imported from a file (anything that is supported by CadQuery and has parameters saved in it)
2. We read a json file with new parameters values ( the parameters are same, but values are different ) 
3. We modify the parameters related to the model in a loop.
4. We export model.

And the another question is: If we can modify the parameters this way, is there a way to do it with skeletal modeling?

jmwright, 2021-02-24:
> Model is imported from a file (anything that is supported by CadQuery and has parameters saved in it)

There are not many model formats that carry parameters with them (maybe IFC?). There are none that we can currently import.

CadQuery scripts are just Python, so you can create a model in a loop, apply whatever parameter changes you want, export the result, then move on to the next cycle of the loop. Python can handle JSON from external files as well.

> If we can modify the parameters this way, is there a way to do it with skeletal modeling

I think I'll need an example of what you mean by "skeletal modelling".

welenely, 2021-02-25:
> There are not many model formats that carry parameters with them (maybe IFC?). There are none that we can currently import.
>
> CadQuery scripts are just Python, so you can create a model in a loop, [...]

Under 'skeletal modelling' I mean the case when we have multiple different objects that share the same parameters.
https://www.youtube.com/watch?v=CMApjCzcG0c&ab_channel=ReproProducts

But your answer is enough for me to understand that I should try a different way, probably with Autodesk Inventor api, thanks for your quick response.

closed 2021-02-25 by welenely

marcus7070, 2021-02-25:
> Under 'skeletal modelling' I mean the case when we have multiple different objects that share the same parameters.

I would refer to that as parametric modelling. 

> But your answer is enough for me to understand that I should try a different way, probably with Autodesk Inventor api, thanks for your quick response.

That would not be my takeaway at all. If you have a look at [our assembly tutorial](https://cadquery.readthedocs.io/en/latest/assy.html), you'll see multiple parts sharing parameters. We don't make a big deal out of models sharing parameters like this because it's part and parcel of being a Python module. It's trivial in CadQuery.

There are numerous ways of sharing parameters between different models. In [this spindle model](https://github.com/marcus7070/spindle-assy-example) I like to use a seperate `dims.py` module to prevent circular dependencies (which back when I was working with commercial CAD were a major problem).