CadQuery/cadquery #660: Trivial cut fails on clean

adam-urbanczyk, 2021-02-24:
```python
result = cq.Workplane("XY").rect(10,10,).extrude(1)\
   .faces(">Z").workplane().center(5., 0).hole(5,clean=True)
```

fedorkotov, 2021-03-04:
Maybe these additional examples can help in your investigation into reasons of the bug.

## example 1
This modification of your example fails with error "Standard_ConstructionError: gp_Dir::Crossed() -result vector has zero norm" if `clean=True` and works if `clean=False`. The same as your original code.
```
result = \
    cq.Workplane("XY")\
      .rect(10,10,)\
      .extrude(1)\
      .faces(">Z")\
      .workplane()\
      .center(5., 0)\
      .circle(5./2)\
      .cutThruAll(clean=True)
```

## example 2
But this code with subtraction of two independently built solids works both with `clean=True` and `clean=False`
```python
result = \
    cq.Workplane("XY")\
      .rect(10,10,)\
      .extrude(1)\
      .cut(
          cq.Workplane("XY")\
            .center(5., 0)\
            .circle(5/2)\
            .extrude(1), clean=True) 
```
[image]

lorenzncode, 2022-09-22:
This is working for me with the occt7.6 branch and also with master (currently ocp-7.5.3.0). I can reproduce the error with ocp-7.5.1beta.

adam-urbanczyk, 2022-09-22:
Solved in OCCT/OCP 7.5.3

closed 2022-09-22 by adam-urbanczyk