CadQuery/cadquery #661: `cutEach` should be `cutEachPoint`

adam-urbanczyk, 2021-02-24:
Issue triggered by this example:

```python
import cadquery as cq

T = 50
wall = 6
diameter = 200
sides = 8
center_hole_dia = 160;

s = cq.Workplane("XY").polygon(sides,diameter).extrude(T)

s = (s.faces(">Z")
     .workplane()
     .hole(center_hole_dia, T-wall)
     .faces('#Z')
     .each(lambda f: cq.Solid.makeCylinder(10,50, f.Center(),-f.normalAt()))
     )

show_object(s.end().cut(s))
```
[image]

the last line is rather terrible, and per analogy to each and `eachPoint` we should have `cutEach` and `cutEachPoint`

marcus7070, 2021-03-01:
Could we get rid of the `cutEach` method entirely if we made `cut(None)` implement a cut of `self.objects` from `self.findSolid(searchStack=False)`? Your example above would become:

```python
s = (s.faces(">Z")
     .workplane()
     .hole(center_hole_dia, T-wall)
     .faces('#Z')
     .each(lambda f: cq.Solid.makeCylinder(10,50, f.Center(),-f.normalAt()))
     .cut()
)
```

We could implement the same behaviour for `union(None)` and intersect as well.

I swear this is your own idea that you told me like a year ago... I should try to find the comment again.

marcus7070, 2021-03-01:
Ah, now I remember. You were talking about adding a subtract keyword argument to extrude. Not the same as my suggestion.

TG9541, 2025-12-30:
@adam-urbanczyk Thanks for your example above - it helped me understand key concepts of CQ a bit better. 

If I use it with a recent version of CadQuery ("cadquery-2.6.dev0-py3.13", bundled with cq-editor), the example above doesn't work: `show_object(s.end().cut(s))` shows nothing. If I, instead, use `show_object(s)` I see what's in the screenshot below. 

From what I take from discussions in other GitHub issues some improvements regarding `cut` have been introduced. 

Since the issue is still open: would you be so kind to explain which changes were introduced to allow subtraction and how to makethe example like the one above work?

Thanks in advance!  

[image]

closed 2025-12-31 by adam-urbanczyk