CadQuery/cadquery #663: section() Results In Empty Object After Crossing Origin

jmwright, 2021-02-24:
The following code does not display anything in CQ-editor, but no error is displayed.
```python
import cadquery as cq
result = cq.Workplane("XY").box(10, 10, 10)
result = result.section(6)
show_object(result)
```
If I change the value passed to `section()` to 5, the slice/section is displayed.

If I run that code from the command line I don't get an error, unless I try to export the resulting object.
```
exporters.export(result, "/home/jwright/Downloads/section_test.svg")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jwright/anaconda3/envs/cadquery/lib/python3.7/site-packages/cadquery/occ_impl/exporters/__init__.py", line 86, in export
    f.write(getSVG(shape, opt))
  File "/home/jwright/anaconda3/envs/cadquery/lib/python3.7/site-packages/cadquery/occ_impl/exporters/svg.py", line 167, in getSVG
    uom = guessUnitOfMeasure(shape)
  File "/home/jwright/anaconda3/envs/cadquery/lib/python3.7/site-packages/cadquery/occ_impl/exporters/svg.py", line 66, in guessUnitOfMeasure
    bb = BoundBox._fromTopoDS(shape.wrapped)
  File "/home/jwright/anaconda3/envs/cadquery/lib/python3.7/site-packages/cadquery/occ_impl/geom.py", line 848, in _fromTopoDS
    return cls(bbox)
  File "/home/jwright/anaconda3/envs/cadquery/lib/python3.7/site-packages/cadquery/occ_impl/geom.py", line 748, in __init__
    XMin, YMin, ZMin, XMax, YMax, ZMax = bb.Get()
OCP.Standard.Standard_ConstructionError: Bnd_Box is void

```
The box is centered on the origin by default, so for some reason `section()` is returning an empty object after "crossing over" the origin.

Ref PR #339

adam-urbanczyk, 2021-02-25:
The cube goes from -5 to 5 and your origin is at 0. So section at h=6 does not produce any objects. 

This should illustrate the situation:
```python
import cadquery as cq
result = cq.Workplane("XY").box(10, 10, 10)
debug(result)
result = result.section(5)
show_object(result)
```
[image]

jmwright, 2021-02-25:
Ok, thanks. I was expecting the section height to be relative to the overall dimensions of the object and not the origin. I understand now.

jmwright, 2021-02-25:
Thanks @adam-urbanczyk

closed 2021-02-25 by jmwright