CadQuery/cadquery #664: Parametrizing a STEP file - Or creating from scratch?

hexad, 2021-02-25:
If i want to parametrize an existing STEP file, is it easy to parametrize it with cadquery?

Consider this shape.
[image]

Consider if i wanted to change the radius of the circle that exists in the top (well a circle does not exist, but it how this compound object was created).

Selecting the surrounding 'wall' and changing the radius, is it something that is within the capabilities of cadquery?

If i do manage to somehow change the radius, i will also have to change the angles of the walls that connect to it.

So i am thinking, for operations like this.... Should i use cadquery to parametrize an existing file? Or should i use cadquery to create it from scratch?

And then if i create it from scratch with cadquery, i could parametrize it more easily by changing parameters to my code....

What do you think based on your experience?

marcus7070, 2021-02-25:
I would say parameterising an existing STEP model like that is not the way to go. It would be very difficult to do and fragile.

There are some high level concepts here I would love to explain, but I'm struggling to get them into words. It's got to do with capturing design intent, which is one of the goals of CadQuery. If you can write your model with CadQuery, changing parameters will be far more robust than importing a STEP.

hexad, 2021-02-25:
Thank you. Got it, if you want to parametrize, create it from scratch. Just one more question

What functionality CADQuery offers (IF it offers) to implement constraints during the 2D Construction stage? I am aware that this can be implemented between topological entities in the 3D Construction stage but what about constraints between sketch entitites (e.g. Line/Arc tangency, Coincident points, etc.)?

jmwright, 2021-02-25:
That functionality is not in place at this time. There are multiple issues related to adding 2D constraints in the next release: https://github.com/CadQuery/cadquery/projects/7

closed 2021-04-06 by marcus7070