CadQuery/cadquery #665: Deprecate `Workplane.findFace`?

marcus7070, 2021-02-25:
I don't think anyone is going to miss it. Shall we mark it deprecated and remove next release?

jmwright, 2021-02-25:
@marcus7070 What's wrong with `findFace`? Is it redundant or something?

marcus7070, 2021-02-25:
Pretty much redundant. It isn't used internally, or in cq-kit or paramak. It didn't have any tests written for it. I just assumed is was an artefact of an old `Workplane.workplane` implementation.

I just went and chased the git blame for it, and I actually see it was added not too long ago by @adam-urbanczyk in 8606e6bb1f4ebdcc2723e1e6059146b18f05d2c6 to support `Workplane.cutThruAll`, and then removed by him in 3348c18889b37b7489468b4a0f59d73494a67389, so I'll leave it up to him if it should be deprecated or not.

To be clear, the only reason I think it should go is to trim 12 lines of code out of the codebase. It doesn't cause any problems or anything.

adam-urbanczyk, 2021-02-25:
Ineed, `cutThruAll` was refactored recently. +1 on deprecating

PR #679 (2021-03-10): Deprecate Workplane.findFace
Will close #665.
closed 2021-03-10 by jmwright