CadQuery/cadquery #675: How to create a rounded box with custom outer surface

zumpchke, 2021-03-04:
Hi,
I would like to create a rounded box with an outer surface that looks like

https://upload.wikimedia.org/wikipedia/commons/e/e3/Openscad_surface_example_x2.png

jmwright, 2021-03-04:
@zumpchke Have a look at [InterpPlate](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=interpplate#cadquery.Workplane.interpPlate)

An example is [here](https://github.com/CadQuery/cadquery/blob/master/examples/Ex101_InterpPlate.py).

jmwright, 2021-03-04:
Also, be careful about pulling code from [here](https://github.com/CadQuery/cadquery/issues/229#issuecomment-561819237) because @bragostin was still working through ideas, but there are some screenshots of the results that are possible with interpPlate.

zumpchke, 2021-03-04:
thanks. So it looks like its possible to generate a complex surface.

Any thoughts on how I would "apply" this to an existing object? I was looking at the sphere example, which had many spikes generated (which are the interpPlate solid). But the parent/master object in this case is a sphere which is well defined mathematically and I wanted to apply to a rounded box.

adam-urbanczyk, 2021-03-04:
Can you paste what you have? Lofting might be an easier modeling strategy for this kind of objects - you'd need generate a series of x-section profiles.

fedorkotov, 2021-03-04:
This is probably not what @zumpchke wanted but something to start with along the lines of @adam-urbanczyk suggestion to use `loft` instead of `InterpPlate`

```python
from math import sin, cos, pi

def f(u,v):
    return sin(2*pi*u)*cos(2*pi*v)

def section(workplane, y, size_x, size_y, depth, scale):
    return\
      workplane\
      .parametricCurve(
          lambda t: (t*size_x, 
                     scale*f(t,y/size_y),
                     0),
          makeWire=False)\
      .line(0,-depth)\
      .line(-size_x,0)\
      .close()\

def sections(
        workplane, 
        size_x, size_y, depth, 
        scale, 
        n_sections):
    
    workplane =\
        section(
            workplane,
            0,
            size_x, size_y, depth,
            scale)
    
    delta_y = size_y/(n_sections-1)
    
    for i in range(1, n_sections):
        y = i * delta_y
        
        workplane =\
            section(
                workplane
                    .workplane(offset=-delta_y),
                y,
                size_x, size_y, depth,
                scale)
    return workplane

result3 =\
  sections(
      cq.Workplane("XZ"),
      5., 5., 2.,
      1.,
      3)\
  .loft()

result5 =\
  sections(
      cq.Workplane("XZ"),
      5., 5., 2.,
      1.,
      5)\
  .loft()  
```

[image]

Be careful with `n_sections`. A loft with 5 sections is very slow to compute on Ubuntu 20.04 which I use now. A loft with 10 sections caused caused my Windows 8.1 machine to freeze completely earlier this day.

A question for @zumpchke . What do you mean by rounded box?
This one?
```python
result = \
    cq.Workplane("XY")\
      .rect(5,5)\
      .extrude(5)\
      .edges()\
      .fillet(1)
```
[image]

Or this one?
```python
result = \
    cq.Workplane("XY")\
      .rect(5,5)\
      .extrude(5)\
      .edges("|Z")\
      .fillet(1)
```

[image]

zumpchke, 2021-03-04:
@fedorkotov the first one

fedorkotov, 2021-03-05:
I don't see an easy way to do it. Hope veteran members of the project can help you.

adam-urbanczyk, 2021-03-05:
@zumpchke  you can remove the top face, create a new one with `interpplate` and then construct a new solid. This is a pretty advanced use case, so it'd be better if you could share more details.

fedorkotov, 2021-03-06:
@zumpchke If you do not intend to manufacture this bumpy rounded box you may have easier time with some "artistic" 3D modeller like Blender than with CAD like CadQuery for this task. Google for Displacement Maps in Blender (https://download.blender.org/documentation/htmlI/ch11s03.html )

zumpchke, 2021-03-08:
@adam-urbanczyk  I'm trying to make acoustic foam. Similar to this https://www.youtube.com/watch?v=Q_viUk4Ce1s
@fedorkotov thanks for the pointer, displacement map looks a good method!

adam-urbanczyk, 2021-03-09:
You don't really need the rounded corners AFAICT. The best solution would be to use: `GeomAPI_PointsToBSplineSurface` but it is not exposed. I'll add an issue to expose a `parametricSurface`.

adam-urbanczyk, 2021-04-01:
@zumpchke  after merging #694 you can do this:
```python
from math import *
from cadquery import *

f = Workplane().parametricSurface(
    lambda u, v: (u, v, 0.5*cos(pi * u) * cos(pi * v)),N=40, tol=1e-6, start=-2.1, stop=2.1
)

res = (
    cq.Workplane().box(4,4,2)
    .split(cq.Workplane().add(f))
    .solids("<Z")
    .edges('|Z').fillet(.5)
)

show_object(res)
```
[image]

closed 2021-06-13 by adam-urbanczyk