CadQuery/cadquery #676: Multiple extrusions/sweep question for gear generation

wentam, 2021-03-04:
I'm attempting to port my gear generator from openscad that can produce arbitrary helix patterns, like this:

[image]
[image]

Getting the general gear profile generation working was simple, and splines are awesome:
[image]

But I'm having a hard time finding a good solution to set up the helix generation.

I would expect to be able twistExtrude multiple times (`p=p.wire().twistExtrude(5,20).wires(">Z").twistExtrude(5,-20)`), but there's something I don't understand because this just errors out. How would I extrude the top face of an object?

I had a little bit of success with sweep and an auxSpine
[image]

Thats very satisfying, but trying to define a herringbone profile with a polyline as the auxSpine just does strange things
[image]

Source is here: https://github.com/wentam/Compound-planetary-gearbox-generator/blob/db55c7ff6a5a3625f614d7179e7d08a9f393230f/test.py

I'd any appreciate ideas on how to best solve this!

fedorkotov, 2021-03-05:
The method you are missing is [toPending(..)](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=toPending#cadquery.Workplane.toPending). But it seems that there is a bug in `twistExtrude(..)`.

This works
```python
result = \
    cq.Workplane('XY')\
      .rect(5,5)\
      .twistExtrude(5,20)\
      .wires(">Z")\
      .toPending()\
      .extrude(5)
```
[image]

But this does not
```python
result = \
    cq.Workplane('XY')\
      .rect(5,5)\
      .twistExtrude(5,20)\
      .wires(">Z")\
      .toPending()\
      .twistExtrude(5,-20)      
      # same with .twistExtrude(-5,-20)
```
Both variants (with distance 5 and -5) extrude backwards and give this shape
[image]

fedorkotov, 2021-03-05:
As a workaround you can probably use [mirror(..)](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=mirror#cadquery.Workplane.mirror)
```python
result = \
    cq.Workplane('XY')\
      .polygon(6, 5.0)\
      .twistExtrude(5,60)\
      .mirror(union=True)
```
[image]

wentam, 2021-03-05:
Thanks! toPending() indeed allows for non-twisted extrusion:

[image]

Mirror doesn't fully achieve what is needed here, as asymmetric profiles need to be possible. It is a valid workaround for symmetric cases though.

I don't fully understand what pending vs non-pending wires are, and I can't find much information about it in the docs. Perhaps a candidate for the 'CadQuery Concepts' section?

fedorkotov, 2021-03-05:
Asymmetric ones you can just glue together from two parts with [union](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=union#cadquery.Workplane.union) like this

```python
result = \
    cq.Workplane('XY')\
      .polygon(6, 5.0)\
      .twistExtrude(-5,60)\
      .union(
          cq.Workplane('XY')\
            .polygon(6, 5.0)\
            .twistExtrude(5,-120))
```
[image]

wentam, 2021-03-05:
Yup! - was just playing with union a bit as a workaround for this.

Thanks for your time! That'll be a good solution for now.

adam-urbanczyk, 2021-03-05:
@wentam great project! As already discussed `toPending` is the way. Don't forget to use `workplane` too:
```python
result = (
    cq.Workplane()
      .rect(5,5)
      .twistExtrude(5,20)
      .wires(">Z").toPending()
      .workplane()
      .twistExtrude(5,-20)
)
```
[image]

@fedorkotov see above, you need to use `workplane` to make `twistExtrude` wokr.

wentam, 2021-03-05:
Aha: workplane does make it behave as expected:
[image]

Makes sense I supposed if extrusion is intended to occur relative to the workplane.

adam-urbanczyk, 2021-03-05:
Regarding `auxSpine`: probably only C1 curves are supported. I did not check it though

wentam, 2021-03-05:
@adam-urbanczyk the auxSpine solution doesn't seem to result in usable gears anyway, as the teeth on the opposite side are of a different shape than the front:

[image]
vs
[image]

adam-urbanczyk, 2021-03-05:
AFAIK: your spine needs to be placed on a cylinder for this to work properly. You can take a look at how `twistExtrude` is actually implemented.

fedorkotov, 2021-03-05:
Thanks a lot @adam-urbanczyk . Issues section of this project is a great supplement for documentation.

adam-urbanczyk, 2021-03-05:
Yes, I try to add the `question` label for this kind of issues. One day we might have better docs and maybe even a FAQ.

Jojain, 2021-03-05:
I am also working on gear and I made one here where you can specify the helix angle you want, (even though helix angle >3° make the filleting not working) It's still kindof WIP but maybe can give you some inspiration.
[https://github.com/CadQuery/cadquery-contrib/blob/master/examples/cylindrical_gear.py]