CadQuery/cadquery #680: Remove deprecated stuff?

marcus7070, 2021-03-08:
We have a few things marked deprecated, which include a message "to be removed in the next release". Here is a timeline, YYYY-MM-DD dates:

* 2020-04-24: 2.0 release
* 2020-07-17: `Workplane.combineSolids` marked deprecated in https://github.com/CadQuery/cadquery/commit/8affc900f7e85cb14295fb8e926dd05114ee3506
* 2020-07-29: `exporters.toString`, `exportShape` and `readAndDeleteFile` marked deprecated in https://github.com/CadQuery/cadquery/commit/e5114e8a1ec4dbf1f5c05082ae89602030704895
* 2020-12-20: 2.1RC1
* 2021-02-04: 2.1 release

Is it time for these 4 to go?

jmwright, 2021-03-08:
@marcus7070 I'm not sure I understand the entries related to `2.0 release`, `2.1RC1` and `2.1 release`. Is there text left over in a version number or changelog?

marcus7070, 2021-03-08:
That's just the dates of the releases. The deprecation message says "next release", so I'm just providing some background info. The "4" I refer to are the methods `combineSolids`, `toString`, `exportShape` and `readAndDeleteFile`, I didn't mean remove all that list.

jmwright, 2021-03-08:
I guess it's a judgement call based on several factors including how heavily used the function was, and how long we think it will reasonably take users to mitigate the impact on their codebases (acknowledging that many codebases will never be updated). The deprecations related to the export interface should be safe to remove, as I think those were used more internally. `combineSolids` might have been used more widely by users, but I think it's probably safe to remove as well unless @adam-urbanczyk knows of a reason to wait.