CadQuery/cadquery #685: fillet2D and chamfer2D in the fluent api

marcus7070, 2021-03-12:
Now that #683 has added 2D fillets and chamfers in the direct API, should we expose it in the fluent API and can we do it neatly?

marcus7070, 2021-03-12:
The major consideration for me is how do you fluently select which vertices to fillet/chamfer? `Workplane.vertices()` can't be used because the wire has not been extruded yet. 

I think we can do a method with a signature like:
```python
class Workplane:
    ...
    def fillet2D(
        self,
        radius: float, selector: Optional[Union[Selector, str]] = None,
        vertices: Optional[List[Vertex]] = None,
    ):
```

Outline of the method:
* consolidate pending edges and wires
* if vertices is not None:
  * fillet those vertices and return
* get all vertices from the wire
* if selector is not None:
  * filter all vertices by selector
* fillet vertices and return

Opinions? @adam-urbanczyk, would this implementation have any effect on #42?

adam-urbanczyk, 2021-03-14:
Before going into API details, do we really need this op in the fluent API? For sure it has to be exposed in the (future) `cq.Sketch` API but I'd like to avoid reimplementing parts of `cq.Sketch`. Do you see yourself using it in the fluent API a lot?

marcus7070, 2021-03-15:
Yeah, after thinking about it for a bit longer, this kind of stuff (selecting and operating on pending wires) will be core sketch API. I don't want to add a fluent API method only for it to be rendered redundant in forseeable future.

closed 2021-03-15 by marcus7070