CadQuery/cadquery #691: Explicitly advise to use miniconda

adam-urbanczyk, 2021-03-15:
And to install it in a portable way

```cmd
bitsadmin /transfer myDownloadJob /download /priority normal https://repo.anaconda.com/miniconda/Miniconda3-latest-Windows-x86_64.exe .
Miniconda3-latest-Windows-x86_64.exe /S /InstallationType=JustMe /D=%MINICONDA_DIRNAME%
set "PATH=%MINICONDA_DIRNAME%;%MINICONDA_DIRNAME%\\Scripts;%PATH%"
activate
```

adam-urbanczyk, 2021-03-15:
Actually I'd consider proposing to use mambaforge - initial tests indicate that it is an order of magnitude faster

jmwright, 2021-03-17:
So we would use the `mamba` command instead of `conda` in most cases? Mamba's documentation claims that it builds environments more reliably than conda, but I have no experience with it.

adam-urbanczyk, 2021-03-17:
Yes, it is super fast but I don't have that much experience with it. Maybe it is too early to recommend.

fedorkotov, 2021-03-17:
+1 for mamba from me.

I tried mamba on top of miniconda on Ubuntu 20.04 today. It is objectively **much** faster. 
My miniconda installation is not new so all packages were already in cache before my experiments.
I created two new environments. Installed cq-editor and cadquery in both (sequentially, not in parallel). I used conda in first environment and mamba in second. Execution time was measured with standard [time](https://man7.org/linux/man-pages/man1/time.1.html) command.

```bash
time conda install -y -c cadquery -c conda-forge cq-editor=master cadquery=master
...
real    1m53.195s
user    1m29.013s
sys     0m4.164s
```
```bash
time mamba install -y -c cadquery -c conda-forge cq-editor=master cadquery=master
...
real    0m20.962s
user    0m17.418s
sys     0m2.050s
```

dcowden, 2021-03-17:
wow, not only is that way faster, but that might even be 'faster than pip' faster. I think a lot of people who like pip probably like the speed, and this would be just as fast, so maybe it would endear that crowd

PR #1362 (2023-08-10): docs - Update installation instructions and advise to use mambaforge
Closes #691

I think many of us have been using mamba install for some time with good results.
closed 2023-08-10 by adam-urbanczyk