CadQuery/cadquery #692: Importing cadquery in python 3.8 on Windows fails

fedorkotov, 2021-03-16:
Cadquery for windows is broken since yesterday for some reason. I first experienced it when Windows CI pipelines for #688 failed. But yesterday I did not have access to Windows machine.

Today on windows machine I got rid of full anaconda I was using before and installed miniconda instead https://repo.anaconda.com/miniconda/Miniconda3-latest-Windows-x86_64.exe

The OS is Windows 8.1

This is how I created conda environment
```
conda create -n cadquery -c cadquery -c conda-forge cq-editor=master cadquery=master
```
And this is pyhton version And my attempt to import cadquery. Sorry for Russian error message. It means `DLL load failed: The specified module could not be found` (the same as in azure pipelines).
```
(cadquery) d:\Fedor\tools\miniconda>python
Python 3.8.8 | packaged by conda-forge | (default, Feb 20 2021, 15:50:08) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery as cq
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\KotovF\miniconda3\envs\cadquery\lib\site-packages\cadquery\__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "C:\Users\KotovF\miniconda3\envs\cadquery\lib\site-packages\cadquery\occ_impl\geom.py", line 5, in <module>
    from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf, gp, gp_XYZ
ImportError: DLL load failed while importing OCP: Не найден указанный модуль.
```

fedorkotov, 2021-03-16:
This is output of `conda list`

[conda_list.txt](https://github.com/CadQuery/cadquery/files/6147248/conda_list.txt)

fedorkotov, 2021-03-16:
I checked with Process Monitor utility from sysinternals and discovered that python.exe searches for `FreeImage.dll` in all folders in path and can't find it when I run `import cadquery as cq`. When I placed the dll downloaded from here http://downloads.sourceforge.net/freeimage/FreeImage3180Win32Win64.zip (x64 version) into `miniconda3\envs\cadquery\` cadquery successfully imported.

pytest on my fork of cadquery also executed without errors.

Does cadquery depend on freeimage directly and not installs it? Or is this a bug in some other dependency down the line?

jmwright, 2021-03-16:
@adam-urbanczyk Has filed a [freeimage bug report](https://github.com/conda-forge/freeimage-feedstock/issues/25). The fix for now is to force an older version: `conda install -c conda-forge freeimage=3.18.0=h33f27b4=0`

fedorkotov, 2021-03-16:
Seems to be resolved. Windows CI pipelines for #688 have completed successfully.
When I created a new conda environment locally on windows machine everything also worked without adding freeimage dll manually

closed 2021-03-16 by fedorkotov

wolfv, 2021-03-19:
Hi @fedorkotov would you mind re-running the CI tests to check if the latest free image builds for windows are fine?

@adam-urbanczyk would be cool if you can close the issue in the freeimage-feedstock if that's the case. Thanks!

fedorkotov, 2021-03-19:
@wolfv windows CI pipelines have just completed without errors in #694 with freeimage 3.18.0

wolfv, 2021-03-19:
Awesome, thanks