CadQuery/cadquery #698: Model works when using rect().extrude(), hangs when using box()

marcus7070, 2021-03-18:
## To Reproduce

In the following, `result0` will work fine, `result1` will not perform the final cut and will not print "finished!".

```python
import cadquery as cq
from cadquery.selectors import AreaNthSelector

result0 = (
    cq.Workplane("XY")
    .rect(20, 20)
    .extrude(10)
    .edges("|Z or <Z")
    .fillet(2)
    .faces(">Z")
    .shell(2)
    .faces(">Z")
    .wires(AreaNthSelector(-1))
    .toPending()
    .workplane()
    .offset2D(-1)
    .extrude(1)
    .faces(">Z[-2]")
    .wires(AreaNthSelector(0))
    .toPending()
    .workplane()
    .cutBlind(2)
)

print(f"rect -> extrude is valid? {result0.val().isValid()}")

result1 = (
    cq.Workplane("XY")
    .box(20, 20, 10, centered=(True, True, False))
    .edges("|Z or <Z")
    .fillet(2)
    .faces(">Z")
    .shell(2)
    .faces(">Z")
    .wires(AreaNthSelector(-1))
    .toPending()
    .workplane()
    .offset2D(-1)
    .extrude(1)
    .faces(">Z[-2]")
    .wires(AreaNthSelector(0))
    .toPending()
    .workplane()
)
print("about to cut")
result1 = (
    result1
    .cutBlind(2)
)
print("finished!")
```

Code is adapted from [here](https://github.com/CadQuery/cadquery/blob/d5ce1321ca100c1d0b23968c09a8a8abb72ab986/examples/Ex026_Case_Seam_Lip.py).

Here is a screenshot of result0:
[image]

## Environment

OS: Ubuntu 20.04

Was CadQuery installed using Conda?: Yes.
<details>
<summary>Output of `conda list` from your active Conda environment:</summary>

[... 221 lines cut ...]
wheel                     0.35.1             pyhd3eb1b0_0  
wrapt                     1.12.1           py38h25fe258_2    conda-forge
wurlitzer                 2.0.1            py38h578d9bd_1    conda-forge
x264                      1!152.20180806       h14c3975_0    conda-forge
xorg-kbproto              1.0.7             h14c3975_1002    conda-forge
xorg-libice               1.0.10               h516909a_0    conda-forge
xorg-libsm                1.2.3             h84519dc_1000    conda-forge
xorg-libx11               1.6.12               h516909a_0    conda-forge
xorg-libxau               1.0.9                h14c3975_0    conda-forge
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxt                1.2.0                h516909a_0    conda-forge
xorg-xextproto            7.3.0             h14c3975_1002    conda-forge
xorg-xproto               7.0.31            h14c3975_1007    conda-forge
xz                        5.2.5                h7b6447c_0  
yaml                      0.2.5                h7b6447c_0  
yapf                      0.30.0             pyh9f0ad1d_0    conda-forge
zeromq                    4.3.3                h58526e2_3    conda-forge
zipp                      3.4.1              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11               h7b6447c_3  
zstd                      1.4.5                h6597ccf_2    conda-forge
```
</details>

Using: Python interpreter

marcus7070, 2021-03-18:
I forgot to check before I submitted this; `clean=False` in the final cut prevents the hang.

fedorkotov, 2021-03-19:
> clean=False in the final cut prevents the hang.

Probably related to #660

lorenzncode, 2026-02-15:
Fixed with #1946.

```py
...
print("finished!")

print(result1.val().isValid())
print(result0.val().Volume())
print(result1.val().Volume())
```

result:
```
rect -> extrude is valid? True
about to cut
finished!
result1.val().isValid()=True
result0.val().Volume()=2637.772773235826
result1.val().Volume()=2637.772773235825

```