CadQuery/cadquery #701: Conflicts on conda install alongside other specifications

noce2, 2021-03-24:
Hi,

I have a simple environment.yml file like:
```.yml
channels:
  - cadquery
dependencies:
  - python>=3.7
  - cadquery=master
  - pip
  - pip:
    - -r requirements.txt
    - -r requirements_dev.txt
```

Conda install (with miniconda on Ubuntu 20.04) fails indicating:
```
UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (pip):

  - cadquery=master -> ipython -> setuptools[version='>=18.5']
  - cadquery=master -> python[version='>=3.7,<3.8.0a0'] -> pip
  - pip -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
  - pip -> python[version='>=3.7']
  - pip -> setuptools -> certifi[version='>=2016.09|>=2016.9.26']
  - pip -> wheel
  - python[version='>=3.7'] -> openssl[version='>=1.1.1j,<1.1.2a'] -> ca-certificates
  - python[version='>=3.7'] -> pip -> setuptools
  - python[version='>=3.7'] -> pip -> wheel

The following specifications were found to be incompatible with each other:
```

My environment.yml configuration seems very simple and in my requirements files I have removed most version pinning to attempt to resolve this issue but it still persists. Anyone able to help with this?

 Are all the dependencies in a cadquery install hard-pinned? Does that mean cadquery can't be conda installed alongside other things in a project?

fedorkotov, 2021-03-24:
I'm not a conda expert by any means but I don't see any pinned packages in cadquery (meta.yml)[https://github.com/CadQuery/cadquery/blob/master/conda/meta.yaml] file except `ocp 7.5`

- Are you creating a new conda environment with this `environment.yml` or updating an existing one? If the later please try creating a new environment and see if the conflict goes away
- Can you share your `requirements.txt` and `requirements_dev.txt` files please?
- Do you have any ideas where this requirement is coming from?
   ```
   pip -> python[version='>=2.7,<2.8.0a0'] -> ca-certificates
   ```

adam-urbanczyk, 2021-03-24:
>     * Can you share your `requirements.txt` and `requirements_dev.txt` files please?

That. If we cannot reproduce, we won't be able to help.

noce2, 2021-03-25:
Thanks for the responses all. I was running a `conda env create` with that environment.yml file. My requirements were:

requirements_dev.txt
```.txt
flake8
tox
coverage
pytest
pytest-runner
```

requirements_dev.txt
```.txt
click>=7.0
loguru
```

As part of my trial & error, I added updated my channels from the OP sow it now looks like:
```
channels:
  - cadquery
  - conda-forge #new
  - defaults        #new
``` 
I realise I hadn't created a .condarc on the OS I was using for this, so I think it was not going to any other channels by default. This might have been me being a noob here 😂 .

fedorkotov, 2021-03-25:
I'm glad you figured it out ) I can confirm that adding `conda-forge` to `channels` section of `environment.yml` was enough for successful conda environment creation on Windows 8.1