CadQuery/cadquery #706: Fillet operations on edges

hexad, 2021-03-30:
Hello community!

I have a question regarding the operation of cadquery.
How are we able to only select the edges that the arrows point and apply a fillet operation? What if these edges are not parallel to the X-Axis?

[image]

Jojain, 2021-03-30:
You can select the >Y face and from there select the >Z edge

fedorkotov, 2021-03-30:
A quick illustration for excellent advice given by @Jojain

```python
def part1(offset):
    return cq.Workplane("ZY")\
    .workplane(offset)\
    .line(10,0)\
    .line(0,15)\
    .line(-5,0)\
    .close()\
    .extrude(0.5, both=True)
 
result =\
    part1(-5)\
    .union(part1(5))\
    .edges(">Y and >Z")\
    .fillet(2)
```
[image]

hexad, 2021-03-30:
Thank you very much!
However, i stumbled upon an issue.
I have this code:

`import cadquery as cq

def part1(offset):
    return cq.Workplane("ZY")\
    .workplane(offset)\
    .line(20,0)\
    .line(-10,15)\
    .line(-10,0)\
    .close()\
    .extrude(0.5, both=True)
 
result =\
    part1(-5)\
    .union(part1(5))\`

That generates this:

[image]

If i try to do the same, thing (essentially, adding these two extra lines in the result variable):
`.edges(">Y and >Z")\
    .fillet(4.5)`

I get that ValueError: Fillets require that edges be selected.

I don't understand why this is happening.

hexad, 2021-03-30:
Now i get it. Thank you very much for your support and help!