CadQuery/cadquery #707: Deleting parent solids- after Boolean Union

hexad, 2021-03-30:
Is it possible to delete the parent solids, after a Boolean Union operation? If yes, then how do we do it, API wise?

fedorkotov, 2021-03-30:
Why do you want to delete them? Are you concerned about memory used by the model?

CadQuery objects are just ordinary python objects obeying all the usual laws. You delete them the same way you would delete any other python object.

Also you don't need to assign each body participating in a boolean operation to a variable beforehand.
See my example in your other issue https://github.com/CadQuery/cadquery/issues/706#issuecomment-810165504

To get more specific advice please provide a minimal working example code demonstrating what you want to do.

fedorkotov, 2021-03-30:
I should probably also mention [show_object](https://cadquery.readthedocs.io/en/latest/cqgi.html) in case you are not familiar with this function. It allows you to chose which of the objects you want to show in CQ-editor and to configure their color and transparency.

hexad, 2021-03-30:
Thank you very much for your support!