CadQuery/cadquery #709: Placing a Circle On Each Face of Cube and Extruding Results In Invalid Solid

jmwright, 2021-03-31:
The following code leads to a mangled tessellation in the 3D view of CQ-editor. Exporting as a STEP file leads to an invalid file according to FreeCAD.

```python
result = (cq.Workplane("XY")
            .box(10.0,10.0,10.0,centered=(True,True,True))
            .faces()
            .circle(1.0,forConstruction=False)
            .extrude(1.0,combine=True,clean=False))
show_object(result)
```
I would have expected that to work at least with the `combine=True`, even if it's not a standard way to use that selector. Here is the result in CQ-editor 0.2.

[image]

And here is the result in the latest master of CQ-editor (OCCT 7.5.1).

[image]

Here it is with the box shown as debug for reference.

[image]

I have tried different values for `combine` and `clean`, and have also tried different `centerOption` values. I get the same results.

marcus7070, 2021-04-01:
There is a `Solid.makeCylinder` method that will result in something slightly different than extruding a circle. Or at least it will make different calls to OCCT.

https://github.com/CadQuery/cadquery/blob/7546866159c21b41b039b1e694942b4445628a91/cadquery/occ_impl/shapes.py#L2519-L2536

Might be worth playing with that and seeing if it's any more reliable. Although I don't have much hope, from just a quick play and investigating the `Shape.geomType()`s returned from the two methods (circle->extrude vs `Solid.makeCylinder`), I can't see any difference.

adam-urbanczyk, 2021-04-01:
AFAICT `sortWiresByBuildOrder` only works with co-planar wires. You'll be better off using `each` here.

fedorkotov, 2021-04-01:
Maybe `extrude` should check if it can work with what is selected and raise an error if the answer is no? 
Or maybe if multiple wires are selected it should automatically use something like `each` internally?

Two more examples

This works
```python
result = (
    cq.Workplane("XY")
    .box(10.0,10.0,10.0,centered=(True,True,True))
    .union(
        cq.Workplane("XY")
        .move(15,0)
        .box(10.0,10.0,10.0,centered=(True,True,True)))
    .faces(">Z")
    .circle(1.0)
    .extrude(1)
    )
```
[image]

And this produces empty result
```python
result = (
    cq.Workplane("XY")
    .box(10.0,10.0,10.0,centered=(True,True,True))
    .union(
        cq.Workplane("XY")
        .move(15,0)
        .box(10.0,10.0,10.0,centered=(True,True,True)))
    .faces("|Z")
    .circle(1.0)
    .extrude(1)
    )

```

adam-urbanczyk, 2021-04-01:
Same as above, in the second example your wires are not coplanar. `sortWiresByBuildOrder` could indeed throw an error.

fedorkotov, 2021-04-01:
@adam-urbanczyk I understood that from your previous post. I posted those examples just to illustrate that extrude can fail for non-obvious reason even in simpler cases than what was originally provided by @jmwright , and that maybe something should be done about it (coplanarity check, or using `each` internally)

NothanUmber, 2021-04-01:
Short feedback from a mere user perspective: Principle of least surprise would be - you can select 1 to N faces, (create workplanes on them), create 2d shapes on top and then extrude or loft them. Every "unless XYZ, then you can't" can be highly confusing.
It is not easy to understand why the example above works for one side only (@fedorkotov's example) but not for all faces (.faces() selector).
So, if such cases could internally be handled that it appears to "just work" from the user's perspective then this would be really great! Just forbidding extruding or lofting circles (or curves) in general would be a severe limitation that would make many projects impossible.

adam-urbanczyk, 2021-04-01:
In the CAD context it is rather unusual to try to extrude at the same time wires that are in different planes and in CQ you cannot "select" multiple workplanes at once. We don't want to forbid something here, just produce an explicit error instead of magically disappearing faces.

NothanUmber, 2021-04-01:
Thanks for the feedback! Thought that if a selector selects more than one entity (e.g. face) then all following operations work on all of these faces, so if I select six faces (e.g. the sides of a cube), create a workplane, a circle and then extrude, then I'd have six cylinders growing perpendicularly out of these six faces? There is still much to learn :) Happy easter!

Jojain, 2021-04-01:
That doesn't work this way. Workplane.polarArray() is the closest thing to what you describe, even though it's quite different.

I think having this kind of functionally of extruding in the normal direction of all the selected faces could be nice.

fedorkotov, 2021-04-02:
@Jojain I agree that this "extrude all wires along their respective normals" would be very nice thing to have. But the patch #710 developed by @adam-urbanczyk will not take away any functionality. It will just provide exception instead of silent failure that garbles or destroys the model. And it does not preclude adding this advanced extrude mode later.

PR #710 (2021-04-03): Check if wires are coplanar
Will resolve #709
closed 2021-04-03 by marcus7070