CadQuery/cadquery #719: How to publish extensions/plugins that are deemed inapropriate for CadQuery fluent API?

fedorkotov, 2021-04-08:
As #713 shows there are cases when something of interest for part of the community technically can be done but is deemed inappropriate for implementation in CadQuery fluent API by the project maintainers. There may be good reasons - don't want to argue with that here.

**What are the proper venues for publishing CadQuery extensions/plugins?** 

1. Of course I can just create my own repo and place the code there. But searchability of that will be 0. The majority of users who would have potentially benefited from the code just won't find it.
2. README mentions 
    - https://github.com/CadQuery/cadquery-contrib seems to be more about usage examples than about extensions
    - https://github.com/michaelgale/cq-kit maybe this is the answer?
3. Or maybe there is some place in base CadQuery where such optional helper/extension methods belong?

Jojain, 2021-04-08:
Well I think the best way to do this is here : [https://github.com/CadQuery/cadquery-plugins](url)

fedorkotov, 2021-04-08:
Thank you. I missed this repo. This is definitely the place. Will submit my code from #713 there when I will have time to rewrite it properly and write some tests

NothanUmber, 2021-04-08:
Thank you for coming up with this, this is imho a very useful thing to have!

martinbudden, 2021-09-02:
@fedorkotov , can you close this issue, since it seems to be resolved.

jmwright, 2021-09-02:
Feel free to reopen if the original request is not covered by the new cadquery-plugins repo.

closed 2021-09-02 by jmwright