CadQuery/cadquery #720: offset2D crashing with revolve

Shimwell, 2021-04-12:
Hi all, CQ is super great thanks so much for making it.

I think I might have stumbled along a small bug while trying to revolve an offset2D object and getting mixed results. Arc works perfectly but tangent and intersection don't return quite what I was hoping for. When chained with the revolve then Python crashes.

## To Reproduce

```python
import cadquery as cq
from jupyter_cadquery.cadquery import show

offset_with_arc = (
    cq.Workplane('XY').
    polyline([
        (212, 27),
        (214, 55),
        (215, 69),
    ]).close().offset2D(10, 'arc').revolve(90)
)

show(offset_with_arc)

offset_with_tangent = (
    cq.Workplane('XY').
    polyline([
        (212, 27),
        (214, 55),
        (215, 69),
    ]).close().offset2D(10, 'tangent').revolve(90)
)

show(offset_with_tangent)

offset_with_intersection = (
    cq.Workplane('XY').
    polyline([
        (212, 27),
        (214, 55),
        (215, 69),
    ]).close().offset2D(10, 'intersection').revolve(90)
)

show(offset_with_intersection)
```

[image]

## Backtrace

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-24-5753e337c1ae> in <module>
     14 
     15 offset_with_tangent = (
---> 16     cq.Workplane('XY').
     17     polyline([
     18         (212, 27),

~/anaconda3/lib/python3.8/site-packages/cadquery/cq.py in revolve(self, angleDegrees, axisStart, axisEnd, combine, clean)
   3076 
   3077         # returns a Solid (or a compound if there were multiple)
-> 3078         r = self._revolve(angleDegrees, axisStart, axisEnd)
   3079         if combine:
   3080             newS = self._combineWithBase(r)

~/anaconda3/lib/python3.8/site-packages/cadquery/cq.py in _revolve(self, angleDegrees, axisStart, axisEnd)
   3503         """
   3504         # Get the wires to be revolved
-> 3505         wireSets = sortWiresByBuildOrder(self.ctx.popPendingWires())
   3506 
[... 489 lines cut ...]
xorg-kbproto              1.0.7             h14c3975_1002    conda-forge
xorg-libice               1.0.10               h516909a_0    conda-forge
xorg-libsm                1.2.3             h84519dc_1000    conda-forge
xorg-libx11               1.6.12               h516909a_0    conda-forge
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
xorg-libxt                1.2.0                h516909a_0    conda-forge
xorg-renderproto          0.11.1            h14c3975_1002    conda-forge
xorg-xextproto            7.3.0             h14c3975_1002    conda-forge
xorg-xproto               7.0.31            h14c3975_1007    conda-forge
xz                        5.2.5                h7b6447c_0  
yaml                      0.2.5                h7b6447c_0  
yapf                      0.30.0                     py_0  
zeromq                    4.3.2                he6710b0_2  
zict                      2.0.0                      py_0  
zipp                      3.1.0                      py_0  
zlib                      1.2.11               h7b6447c_3  
zope                      1.0                      py38_1  
zope.event                4.4                      py38_0  
zope.interface            4.7.1            py38h7b6447c_0  
zstd                      1.4.5                h0b5b093_0  
```

Using: 
Also using Jupyter-CadQuery

fedorkotov, 2021-04-12:
Points `(212, 27), (214, 55), (215, 69)`, belong to a straight line. Maybe this is the problem?
```
>>> x0=np.array([212.0,27.0])
>>> x1=np.array([214.0,55.0])
>>> x2=np.array([215.0,69.0])
>>> x1-x0
array([ 2., 28.])
>>> x2-x1
array([ 1., 14.])
```

Shimwell, 2021-04-12:
Ah understood, thanks very much for taking a look and solving.

closed 2021-04-12 by Shimwell

fedorkotov, 2021-04-12:
With degeneracy removed two of your examples work (arc and intersection) but `offset2D(10, 'tangent')` fails (produces empty geometry). 
CadQuery docs do not explain what 'tangent' means. Maybe it has some additional requirements for input geometry.

```python
offset_with_arc = (
    cq.Workplane('XY').
    polyline([
        (212, 27),
        (100, 55),
        (215, 69),
    ]).close().offset2D(10, 'arc').revolve(90)
)
```
[image] 

```python
offset_with_intersection = (
    cq.Workplane('XY').
    polyline([
        (212, 27),
        (100, 55),
        (215, 69),
    ]).close().offset2D(10, 'intersection').revolve(90)
)
```
[image]

Shimwell, 2021-04-12:
The minimal example I provided was just a few points from a larger list of points 

So I'm going to work through these and remove any that are degenerate. 

I guess I have mid points in the list which don't really need to be there for a polyline.

Thanks for your help