CadQuery/cadquery #721: kind='tangent' parameter of offset2D not documented

fedorkotov, 2021-04-12:
Related to #720.
What does `kind='tangent'` argument of [Workplane.offset2D(..)](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=offset2d#cadquery.Workplane.offset2D) and [Wire.offset2D(..)](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=offset2d#cadquery.Wire.offset2D) mean? Does this method of producing offset wires have some non-obvious requirements for input geometry? Can't find any info documentation or source code.

jmwright, 2021-04-16:
[BRepOffsetAPI_MakeOffset().Init()](https://github.com/CadQuery/cadquery/blob/39e60f6e481092c5d5542f370d01c17502efb4a7/cadquery/occ_impl/shapes.py#L1881) has a [JoinType](https://dev.opencascade.org/doc/refman/html/_geom_abs___join_type_8hxx.html#ad03b8c9363e1c78dc022d2f253273c61) parameter, and tangent is a member of that enum. That second link is all the information I've been able to find on it in the OCCT documentation.

fedorkotov, 2021-04-17:
It seems to me that this variant is not implemented at all. This [search in Open Cascade community Edition sources](https://github.com/tpaviot/oce/search?q=GeomAbs_Tangent) yielded two matches: enum definition and some text parser. Can anyone with access to [the official Open Cascade repository](https://dev.opencascade.org/resources/git_repository) and understanding of Open Cascade architecture both of which I don't have confirm this?

fedorkotov, 2021-04-24:
I was wrong about "not implemented". But what exactly does it do is not clear

This works
```python
r1 = cq.Workplane().polygon(4, 1)
r2 = cq.Workplane().polygon(4, 1).offset2D(0.5, "tangent")
```
[image]

And this produces empty result
```python
result = cq.Workplane().polygon(3, 1).offset2D(0.5, "tangent")
```

`intersection` and `arc` work for the same triangle
```python
r1 = cq.Workplane().polygon(3, 1)
r2 = cq.Workplane().polygon(3, 1).offset2D(0.5, "intersection")
r3 = cq.Workplane().polygon(3, 1).offset2D(0.5, "arc")
```
[image]