CadQuery/cadquery #728: documentation: examples page takes long time to load

roipoussiere, 2021-04-16:
I use an old computer and each time I open the examples page, my browser freeze then display an error about a too long loading time. I must look at the documentation source to actually read the provided examples, which is not really convenient.

This is probably due to the rendering of those many svg images.

I would suggest to split this page in different thematic sections. This will also add some structure and probably make the documentation easier to read.

fedorkotov, 2021-04-16:
I can confirm that [examples](https://cadquery.readthedocs.io/en/latest/examples.html) page takes a lot of time to load. My browser does not completely freeze though. **Interactive 3D views in sphinx docs are awesome but are they really required to understand what each example is about?** I think each of them should be replaced with one or two static 2D images. After that 3D views can be moved to individual pages with links from examples page.

fedorkotov, 2021-04-16:
Or otherwise we can have examples list page with each example on separate page. This will require less work to do (no need for additional images)

roipoussiere, 2021-04-16:
I didn't know there was interactive 3D views. :D Yeah replacing them with static svg could solve the problem.

marcus7070, 2021-04-16:
There are a few options to improve this. We can add a lazy load to sphinxcadquery (the sphinx extension that generates and embeds the models) like https://modelviewer.dev/examples/loading/index.html#customizeLoad
We can move each example to it's own sub-page and have an index page (which I would prefer even without the performance issues, the examples page is so long it's easy to miss-scroll and lose your place).

We need to update sphinxcadquery to handle assemblies anyway, so I'll try to improve this at the same time.

Related to #441.

adam-urbanczyk, 2021-04-16:
I propose to not touch sphinxcadquery yet - I'm quite optimistic regarding using vtk.js for visualization in the docs so I'd propose to wait till it is investigated. See #281 for updates.

BTW: I actually find the current layout of the examples nice #bikeshedding

closed 2021-04-16 by adam-urbanczyk