CadQuery/cadquery #730: svg export: ability to define the camera position and zoom

roipoussiere, 2021-04-16:
Let's start with this code:

```py
import cadquery as cq
from cadquery.cq import Workplane as WP

def fig(wp):
    cube1 = WP(wp).box(5, 5, 5)
    rod1 = WP(wp).circle(1).extrude(15)
    return cube1.union(rod1)

svg_opts = {'height': 100, 'showHidden': False, 'showAxes': False}
cq.exporters.export(fig('XY'), 'fig1.svg', opt=svg_opts)
cq.exporters.export(fig('XZ'), 'fig2.svg', opt=svg_opts)
```

[image][image]

These images share the same height. But what I need is to have the same box size and position. So instead of defining svg `width` and `height` in svg export options, I would like to set the camera position and zoom.

My current hack is to add a "background plate", like this:

```py
bg = WP('ZX').transformed(offset=V(-3, 0, -5)).box(1, 17, 25)`)
```

[image][image]

Then remove this plate in Inkscape, but it's a little bit tedious.

jmwright, 2021-04-16:
> I would like to set the camera position and zoom.

For camera position (setting aside zoom for now), `projectDir` does not work for your use case? https://cadquery.readthedocs.io/en/latest/importexport.html#exporting-svg

roipoussiere, 2021-04-16:
> For camera position (setting aside zoom for now), `projectDir` does not work for your use case?

No, event when using `projectDir` the boxes don't have the same size.