CadQuery/cadquery #731: svg export: axes are not displayed when setting the projection dir

roipoussiere, 2021-04-16:
Here is a code that render a simple cube in a svg file:

```py
import cadquery.cq
from cadquery.cq import Workplane as WP

figure = WP('XY').box(3, 3, 3)
cq.exporters.export(figure, 'figure.svg')
```

[image]

When I update the projection dir, the axes disappear:

```py
cq.exporters.export(figure, 'figure.svg', opt = {
    'projectionDir': (0.3, -0.2, 0.1)
})
```

[image]

It's maybe related to #599.

jmwright, 2021-04-16:
Correct on it being related to #599 because the axes are hidden on purpose when setting the projection dir because they're not dynamic. Otherwise the axes indicator would not match the projection direction, causing confusion.

jmwright, 2021-04-16:
BTW, this behavior is documented in the `showAxes` bullet point in the docs: https://cadquery.readthedocs.io/en/latest/importexport.html#exporting-svg

roipoussiere, 2021-04-16:
Ok so I close this since it will be fixed in #599.

closed 2021-04-16 by roipoussiere