CadQuery/cadquery #732: What is the default measurement unit and how to config?

huydhoang, 2021-04-18:
Hi, I'm a newbie in CAD and found this project amazing because it supports Python, seems lightweight and powerful. However, as I read though the [online docs](https://cadquery.readthedocs.io/en/latest/) and [README.md](https://github.com/CadQuery/cadquery/blob/master/README.md), I couldn't locate any reference to the default measurement unit in CadQuery. Is it millimeters? Is there any command to configure this setting if it wasn't mm?

Jojain, 2021-04-18:
I believe if you export a cadquery model and open it with another software the measurement units will be assumed as mm. However there isn't really a measurement unit consideration in cadquery. If you want your model in a specific unit just multiply or divide your values by the right factor.

Most of 3D software will allow you to scale your model so units aren't important as long as you use a consistent unit system across your model

huydhoang, 2021-04-18:
Thank you, this is really helpful for a beginner like me.

closed 2021-04-18 by huydhoang