CadQuery/cadquery #736: Export Assembly as TSJ

davidvpspace, 2021-04-20:
Hi there!

I have an assembly created by multiple parts in cadquery. I would like to export it as json mesh format to later apply some textures to the different parts. With cadquery, the assembly cannot be exported directly as .tjs since only .step and .xml is available.

I have tried a workaround to convert the assembly to a single piece by and then export it to .tjs

`
assy = assy.toCompound()
cq.exporters.export(assy, 'design1.json', tolerance=0.01, angularTolerance=0.1, exportType=exporters.ExportTypes.TJS)
`

However, I get the following error:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-7f7c910c646a> in <module>
----> 1 assy.toCompound()

AttributeError: 'Assembly' object has no attribute 'toCompound'
```

I have installed cadquery using the master branch so I should be able to use the attribute toCompound(). Do you have an idea of what could be the source of this? 

Also, is it in mind to add to cadquery a feature to specify material type to different parts of an assembly? 

Thank you !

davidvpspace, 2021-04-20:
I have also try the solution from issue #629, but I get the following error:

`assy_solid = generate_solid_from_assembly(assy)`

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-7-bf842f15567a> in <module>
----> 1 assy_solid = generate_solid_from_assembly(thruster)

<ipython-input-6-5f4c196566c5> in generate_solid_from_assembly(assy, fuse, name_filter_re)
      6             continue
      7 
----> 8         shape = val.obj.val()
      9 
     10         # get location

AttributeError: 'NoneType' object has no attribute 'val'
```

davidvpspace, 2021-04-20:
I have also try the solution from issue #629, but I get the following error:

`assy_solid = generate_solid_from_assembly(assy)`

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-7-bf842f15567a> in <module>
----> 1 assy_solid = generate_solid_from_assembly(assy)

<ipython-input-6-5f4c196566c5> in generate_solid_from_assembly(assy, fuse, name_filter_re)
      6             continue
      7 
----> 8         shape = val.obj.val()
      9 
     10         # get location

AttributeError: 'NoneType' object has no attribute 'val'
```

fedorkotov, 2021-04-20:
Please provide a minimum (non)working example including the code that creates some assembly so that we can reproduce the problem.

Jojain, 2021-04-20:
When did you installed cadquery ? The toCompound method has been merged yesterday, you might want to update your installation first to see if it solves the problem.

davidvpspace, 2021-04-20:
I installed it a couple weeks ago. I tried to update it this morning. Maybe I should try again, is it done by doing:?

`conda install -c cadquery -c conda-forge cadquery=master`

davidvpspace, 2021-04-20:
> Please provide a minimum (non)working example including the code that creates some assembly so that we can reproduce the problem.

This is the complete working code, but that is unable to create a single solid part:

```
import cadquery as cq
import re
import time
from typing import Optional

class CADCreator():
    
    def __init__(self, var=None):
        self.pmuDictInit = {"height":100, "width":100, "thickness":25, "attach_height":12,"chamfer":11, "n_emitter":8 } if var is None else var
        self.pmuDictInit["fillet"] = 2
        self.emitterDictInit = {"em_width":22, "gr_width":20, "depth":5, "t_grid":0.5, "hole_number":11, "spacing_grid_side_factor":8, "spacing_grid_factor":2}
        self.connector = {"spacing":2.54, "pin":0.635, "pin_grid":(26,4),"thickness":10}
        self.logoDictInit = {"t_logo":0.01, "side_logo":self.emitterDictInit["gr_width"]/(3+0.1)}
        self.cwidth = (self.emitterDictInit["em_width"]+self.emitterDictInit["gr_width"])/2
        self.d_hole = self.cwidth/(2*self.emitterDictInit["spacing_grid_side_factor"]+self.emitterDictInit["hole_number"]+self.emitterDictInit["spacing_grid_factor"]*(self.emitterDictInit["hole_number"]-1))
        self.sw = (self.pmuDictInit["width"]-3*self.emitterDictInit["em_width"])/8
        self.sh = (self.pmuDictInit["height"]-3*self.emitterDictInit["em_width"])/8
        self.connector["width"] = self.connector["spacing"]*self.connector["pin_grid"][0]
        self.connector["height"] = self.connector["spacing"]*self.connector["pin_grid"][1]
        self.pcbAttachPoints = [(-37.505,42.865),(36.195,42.865),(-40.005,-42.865),(40.005,-42.865)]
        self.pc104 = [90.2,95.9,15]
        
        self.emit_points_all = {"1":(-2*self.emitterDictInit["em_width"]/2-self.sw,-2*self.emitterDictInit["em_width"]/2-self.sh),"2":(0,-2*self.emitterDictInit["em_width"]/2-self.sh),\
                            "3":(2*self.emitterDictInit["em_width"]/2+self.sw,-2*self.emitterDictInit["em_width"]/2-self.sh),"4":(-2*self.emitterDictInit["em_width"]/2-self.sw,0),\
                            "5":(2*self.emitterDictInit["em_width"]/2+self.sw,0),"6":(-2*self.emitterDictInit["em_width"]/2-self.sw,2*self.emitterDictInit["em_width"]/2+self.sh),\
                            "7":(0,2*self.emitterDictInit["em_width"]/2+self.sh),"8":(2*self.emitterDictInit["em_width"]/2+self.sw,2*self.emitterDictInit["em_width"]/2+self.sh)}
            
        self.logo_points_purple = [(0,self.emitterDictInit["gr_width"]-self.logoDictInit["side_logo"]*2),(self.emitterDictInit["gr_width"]-self.logoDictInit["side_logo"]*2,
                                    self.emitterDictInit["gr_width"]-self.logoDictInit["side_logo"]*2), (-self.emitterDictInit["gr_width"]+self.logoDictInit["side_logo"]*2,0),\
                                   (self.emitterDictInit["gr_width"]-self.logoDictInit["side_logo"]*2,0),(-self.emitterDictInit["gr_width"]+self.logoDictInit["side_logo"]*2,\
                                    -self.emitterDictInit["gr_width"]+self.logoDictInit["side_logo"]*2),(0,-self.emitterDictInit["gr_width"]+self.logoDictInit["side_logo"]*2),\
                                    (self.emitterDictInit["gr_width"]-self.logoDictInit["side_logo"]*2,-self.emitterDictInit["gr_width"]+self.logoDictInit["side_logo"]*2)]
            
        self.logo_point_green =[(-self.emitterDictInit["gr_width"]+self.logoDictInit["side_logo"]*2,self.emitterDictInit["gr_width"]-self.logoDictInit["side_logo"]*2)]

        self.grid_holes = []
        for i in range(0,self.emitterDictInit["hole_number"]):
            for j in range(0,self.emitterDictInit["hole_number"]):
                if j%2==0:
                    self.grid_holes.append((self.cwidth/2-(self.emitterDictInit["spacing_grid_side_factor"]+0.5)*self.d_hole-self.d_hole*(self.emitterDictInit["spacing_grid_factor"]+1)*i,\
                                        -self.cwidth/2+(self.emitterDictInit["spacing_grid_side_factor"]+0.5)*self.d_hole+self.d_hole*(self.emitterDictInit["spacing_grid_factor"]+1)*j))
        for i in range(0,self.emitterDictInit["hole_number"]+1):
            for j in range(0,self.emitterDictInit["hole_number"]):
                if j%2>0:
                    self.grid_holes.append((self.cwidth/2-(self.emitterDictInit["spacing_grid_side_factor"]-(self.emitterDictInit["spacing_grid_factor"]+1)/2+0.5)*self.d_hole-self.d_hole*(self.emitterDictInit["spacing_grid_factor"]+1)*i,\
                                        -self.cwidth/2+(self.emitterDictInit["spacing_grid_side_factor"]+0.5)*self.d_hole+self.d_hole*(self.emitterDictInit["spacing_grid_factor"]+1)*j))
        if self.pmuDictInit["n_emitter"]==8:
            self.emit_points = [self.emit_points_all["1"],self.emit_points_all["2"],self.emit_points_all["3"],self.emit_points_all["4"],self.emit_points_all["5"],\
                                self.emit_points_all["6"],self.emit_points_all["7"],self.emit_points_all["8"]]
        elif self.pmuDictInit["n_emitter"]==6:
            self.emit_points = [self.emit_points_all["1"],self.emit_points_all["2"],self.emit_points_all["3"],self.emit_points_all["6"],self.emit_points_all["7"],self.emit_points_all["8"]]
            
        elif self.pmuDictInit["n_emitter"]==4:
            self.emit_points = [self.emit_points_all["1"],self.emit_points_all["3"],self.emit_points_all["6"],self.emit_points_all["8"]]
        self.pinPoints = []
        for i in range(0,self.connector["pin_grid"][0]):
[... cut ...]
        result.append(shape.located(loc))

    result = cq.Compound.makeCompound(result) 

    if fuse:
        result = result.fuse()

    return result

start_time = time.time()
var = {"height":85, "width":86, "thickness":30, "attach_height":5,"chamfer":12,"n_emitter":8}

test = CADCreator(var)
test.buildPMU()
test.buildEmitters()
test.buildLogo()
test.buildEbox()
test.buildAT()
thruster = test.assemblyThruster()
test.exportSTEP()
show_object(thruster)

assy_solid = generate_solid_from_assembly(thruster)
```

davidvpspace, 2021-04-20:
> I have also try the solution from issue #629, but I get the following error:
>
> `assy_solid = generate_solid_from_assembly(assy)`
>
> ``` [...]

I was able to solve this. I was creating an empty assembly and then adding parts. The assembly was being created properly but when I tried to run the generate_solid_from_assembly function didnt work

```
assy = cq.Assembly()
assy.add(part1, name="part1")
assy.add(part2, name="part2")
```

fedorkotov, 2021-04-20:
> I was able to solve this.

I'm glad that you have managed to solve this problem yourself.

> I was creating an empty assembly and then adding parts. The assembly was being created properly but when I tried to run the generate_solid_from_assembly function didnt work

Previously you posted and then deleted some huge piece of code depending on a DXF that you did not provide. Now after you have deleted that post your comment does not really explain what happened. Nobody except your knows what was inside `generate_solid_from_assembly ` function.
Please consider adding a short summary for the benefit of others who will come from google to this issue if you have time for this.

davidvpspace, 2021-04-20:
> I'm glad that you have managed to solve this problem yourself.
>
>
> Previously you posted and then deleted some huge piece of code depending on a DXF that you did not provide. Now after you have [...]

Yes,  I realized that the whole code was too much for the simple question I was asking. The code inside generate_solid_from_assembly is the same as in #629 

```
def generate_solid_from_assembly(assy, fuse=True, name_filter_re=None):
    result = []
    
    for name,val in assy.traverse():
        if name_filter_re and not re.match(name_filter_re, name):
            continue
        
        shape = val.obj.val()

        # get location
        loc = val.loc
        current = val
        while current.parent:
            loc = current.parent.loc*loc
            current = current.parent
        
        result.append(shape.located(loc))

    result = cq.Compound.makeCompound(result) 

    if fuse:
        result = result.fuse()

    return result
```

I was giving the function an assembly that was initially created empty and then adding the parts:

```
assy = cq.Assembly()
assy.add(workplane1, name="part1")
assy.add(workplane2, name="part2")
assy_solid = generate_solid_from_assembly(assy)
```

This was an issue as the method val() was not recognized. On the other hand if I do:

```
assy = cq.Assembly(workplane1, name="part1")
assy.add(workplane2, name="part2")
assy_solid = generate_solid_from_assembly(assy)
```

works correctly.

To close this issue, what is the command line that I should write in the conda prompt to update cadquery to include the method toCompound(). This would replace the whole function. Thx!

fedorkotov, 2021-04-20:
> To close this issue, what is the command line that I should write in the conda prompt to update cadquery to include the method toCompound(). This would replace the whole function. Thx!

As @Jojain said `toCompound()` was merged yesterday. So you should install the latest master from this repo.

Maybe this is not the correct way but I usually just delete conda environment where my copy of CadQuery is installed and reinstall everything as described in CadQuery [README](https://github.com/CadQuery/cadquery#getting-started) or with mamba (see #691). This avoids all possible version conflicts and ensures latest possible versions of all dependencies are installed.

There is also `conda update --all` (see https://stackoverflow.com/questions/38972052/anaconda-update-all-possible-packages for example).

jmwright, 2021-04-20:
@davidvpspace You can try the following method, but what @fedorkotov has suggested about removing and re-creating the environment is the safest method. We have had multiple users end up with weird version conflicts when upgrading packages within environments.

```
conda remove cadquery
conda clean --all --force-pkgs-dirs
conda install --force-reinstall -c conda-forge -c cadquery=master
```

adam-urbanczyk, 2021-04-20:
Please wait till azure on master is green (should happen soon).

davidvpspace, 2021-04-21:
@jmwright I have done what you proposed of removing cadquery and reinstalling it, but now I get the following error. I have tried several time to delete all the files and installing it again, but the same happens:

```
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\david\anaconda3\envs\cadquery\lib\site-packages\cadquery\__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "C:\Users\david\anaconda3\envs\cadquery\lib\site-packages\cadquery\occ_impl\geom.py", line 5, in <module>
    from OCP.gp import (
ImportError: DLL load failed while importing OCP: The specified module could not be found.
```

jmwright, 2021-04-21:
CadQuery CI is green again, so you should be safe to try this now. Did you use the following steps to remove and recreate the environment?
```bash
conda deactivate # if you are in the cadquery environment
conda env remove -n cadquery # Should say "Removing all packages" and not give an error
conda create -n cadquery python=3.8 # I personally like to pin the version of Python
conda activate cadquery
conda install -c conda-forge -c cadquery cadquery=master
```
Then try `import cadquery`. If the steps above do not work, please post what OS you're running and what version of Anaconda/Miniconda/Mamba you are running.

davidvpspace, 2021-04-21:
Now it works! (with the steps above) Thanks for your help!

closed 2021-04-21 by davidvpspace