CadQuery/cadquery #740: VRML Export - File extension

BrandonEscamilla, 2021-04-22:
Hi Everyone, 

I was trying to export a VRML file and I noticed that the file extension cadquery uses is .vrml but in other apps I've used VRML files with the filename extension as .wrl, I am wondering if this is intended, if so, Is there something I'm missing?

adam-urbanczyk, 2021-04-22:
I don't have much experience with vrml so this might be non-standard. It is easy to change and you can specify any extension you like and force VRML output using `exportType='VRML`.

jmwright, 2022-04-21:
You can set the file extension to whatever you want. Example [here](https://gitlab.com/jmwright/kicad-packages3D-generator/-/blob/master/Battery/main_generator.py#L173).

lorenzncode, 2022-04-28:
I tested with 3 different viewers today.  All recognized files with .wrl extension.

meshlab did not recognize VRML files with .vrml extension (the other two were dune, freewrl).

I'd suggest to reopen as an enhancement.  The enhancement would be for `exporters.export` and `assembly.save` to check for "WRL" extension (in addition to "VRML").