CadQuery/cadquery #741: model may be merged on multiple svg export

roipoussiere, 2021-04-23:
Here is a code that render 2 figures and export them in separated svgs:

```py
import cadquery as cq
from cadquery.cq import Workplane as WP
from cadquery.cq import Vector as V

cube1 = WP('YZ').move(1, 0).box(2, 2, 2)
cube2 = WP('XY').box(6, 3, 3)

def fig1():
    return cube2.union(WP('XY').transformed(rotate=V(45, 0, 0)).box(6, 3, 3))

def fig2():
    return cube2.union(WP('XY').transformed(rotate=V(0, 45, 0)).box(6, 3, 3))

def export(fig, fig_name):
    cq.exporters.export(cube1.add(fig), fig_name, opt = {'showHidden': False})

export(fig1(), 'fig1.svg')
export(fig2(), 'fig2.svg')
```

The `fig1.svg` file is rendered as expected:

[image]

But in `fig2.svg`, fig1 and fig2 are merge together, instead of jut fig2.

[image]

adam-urbanczyk, 2021-04-23:
That has nothing to do with export and everything with add:
https://github.com/CadQuery/cadquery/blob/c6c1956429a35ae30a67f62984c1f6f4b979690c/cadquery/cq.py#L458
`add` modifies current Wokrplane in-place.

roipoussiere, 2021-04-23:
Ok so is it related to #593 and will be fixed with #449?

adam-urbanczyk, 2021-04-23:
#449 will fixes disjoint overlapping solids,it has no relation to what `add` does and which side-effects it has.