CadQuery/cadquery #742: Extract wire properties (coordinates and spline type)

Andy-Valentine, 2021-04-24:
I am working on a specialized slicer which would create micropatterns on walls. Modelling these micropatterns in very complex 3D objects which I work with is not really feasible (due to too many polygons and resulting near infinite calculation time), so I decided to slice the 'smooth' object and modify the wires in each layer instead and then export the patterned wires layer-by-layer to 3D printer.

So, here is the script that can micropattern any given polyline:

```

'dropping points on each line segment'
def points(start,end,step):
    dist=np.sqrt((end[1]-start[1])*(end[1]-start[1])+(end[0]-start[0])*(end[0]-start[0]))
    Nst=int(dist/step)+1
    #print(dist,Nst)
    x=np.linspace(start[0], end[0], Nst)
    y=np.linspace(start[1], end[1], Nst)
    points=zip(x,y)
    return list(points)

'applying a micropattern at each point'
def micropattern(coordinates,angle,step):
    x=coordinates[0]
    y=coordinates[1]
    x1=x-amp*cos(angle)
    y1=y+amp*sin(angle)
    x2=x-amp*cos(angle)+step*1/4*sin(angle)
    y2=y+amp*sin(angle)+step*1/4*cos(angle)
    x3=x+(amp)*cos(angle)+step*2/4*sin(angle)
    y3=y-(amp)*sin(angle)+step*2/4*cos(angle)
    x4=x+(amp)*cos(angle)+step*3/4*sin(angle)
    y4=y-(amp)*sin(angle)+step*3/4*cos(angle)
    return [x1,y1,x2,y2,x3,y3,x4,y4]

pts=[[1,1],[5,20],[9,21],[4,0],[-7,-3]]
step=0.5
amp=0.4

path=[]
for i in range(len(pts)-1):
    segment=points(pts[i],pts[i+1],step)
    path.append(segment)

'patterning path'
PTS=[]
for j in range(len(path)):
    for i in range(len(path[j])):
        if i<len(path[j])-1:
            angle=np.arctan2((path[j][i+1][0]-path[j][i][0]),(path[j][i+1][1]-path[j][i][1]))
            #print(degrees(angle))
        coords=micropattern(path[j][i],angle,step)
        for m in range(int((len(coords)+1)/2)):
            PTS.append([coords[2*m],coords[2*m+1]])

'Drawing lines'
line=cq.Workplane().polyline(pts)
wigline=cq.Workplane().polyline(PTS)
show_object(line,options=dict(alpha=0,color='black'))
show_object(wigline,options=dict(alpha=0,color='orange'))

```
It looks like this: black line is the original wire, orange is the desired end result:

[image]

Now I take a 3D object, slice it and extract the section wire (blue):
```
base=cq.Workplane().box(5,7,12)
sphr=cq.Workplane().sphere(5).translate((0,3,0))

part=base.union(sphr)

wireslice=part.section(1).wires()

show_object(wireslice,options=dict(alpha=0,color='blue'))
show_object(part,options=dict(alpha=0.5,color='yellow'))

```
[image]

The question:
I couldn't find a way to extract cross-sectional wire coordinates as a list to feed it to the 'micropattern' function. I could extract individual vertices using  

```
wireslice.vertices().vals()[1].Center()
```
but this seems like a tedious task, and I lose spline information. There must be a better way.

What would be the smart way of extracting points from these cross-sectional wires?  
Or maybe I should just place points on a wire with a given step size and then use those points? 
Is there a way to apply pushPoints() along the wire?

fedorkotov, 2021-04-24:
I think this is related to #675. @zumpchke tried to create a rounded box like the one below covered with wavy patterns
```python
result = \
    cq.Workplane("XY")\
      .rect(5,5)\
      .extrude(5)\
      .edges()\
      .fillet(1)
```
[image]

And even after @adam-urbanczyk implemented  #694 there seem to be no easy way to apply an arbitrary pattern to a complex object like yours in CadQuery. 
The main difficulty is blending corners
[image]

I would try to use "artistic" mesh modeller like Blender for this if I were you. Maybe create base "smooth" model in CadQuery, export it to something Blender understands and apply a Displacement Map there (https://download.blender.org/documentation/htmlI/ch11s03.html ).

fedorkotov, 2021-04-24:
Also depending on your geometry and pattern maybe you can do something like what this guy does in FreeCAD  https://www.youtube.com/watch?v=Yyffu0dY8fc
- create base model
- create a rectangular block filled with a pattern (honeycomb in the video) and add it to model
- "shave" the pattern with offseted model surface

adam-urbanczyk, 2021-04-24:
You are probably interested in `cq.Wire.positions(...)`.

Andy-Valentine, 2021-04-24:
Thanks for your comments, the `cq.Wire.positions(...)`  seems to work well.

regarding 'artistic' software, it usually works well, but scales disastrously. I will have around 0.1-1 million of such patterning features on the part surface, and not on the simple flat surface, but on a complex-shape object with multiple intersections and booleans etc. 

So no Blender etc., CQ is the best :)

closed 2021-04-24 by Andy-Valentine