CadQuery/cadquery #743: DOI Badge Is Broken

jmwright, 2021-04-24:
I noticed that the DOI badge in the readme was just displaying text instead of an image, and looked into it. It looks like the links for the badge SVG and the URL the badge points to have changed.

Old DOI Badge Markdown:
```
[![DOI](https://zenodo.org/badge/155099400.svg)](https://zenodo.org/badge/latestdoi/155099400)
```

New DOI Badge Markdown:
```
[![DOI](https://zenodo.org/badge/DOI/10.5281/zenodo.4498634.svg)](https://doi.org/10.5281/zenodo.4498634)
```
I clicked on the DOI badge in the right sidebar [here](https://zenodo.org/record/4498634) to get the new information. I can change it, but wasn't involved in adding the DOI badge and I wanted to make sure I wasn't missing something.

fedorkotov, 2021-04-28:
Is it really fixed? This is what I see (Firefox 88.0 64-bit, Windows 8.1)
[image]

jmwright, 2021-04-28:
@fedorkotov  I wasn't paying attention and didn't get that change committed with the other PR. I've fixed it and CI is running on the merge right now. If it's not fixed after CI has finished, I'll edit directly on GitHub so that I can use the preview.

jmwright, 2021-04-28:
So the link is updated in the badge now, but it seems we're not alone in having the badge refuse to display.

https://github.com/zenodo/zenodo/issues/2165

There are a couple of odd things in their generated SVG file. I may experiment with that a little bit.

fedorkotov, 2021-04-28:
> There are a couple of odd things in their generated SVG file

It takes 26 seconds to load from where I am (Moscow, Russia).
After it is finally loaded it looks ok
[image]
```xml
<svg xmlns="http://www.w3.org/2000/svg"
     width="186" height="20">
        <linearGradient id="b" x2="0" y2="100%">
            <stop offset="0" stop-color="#bbb" stop-opacity=".1"/>
            <stop offset="1" stop-opacity=".1"/>
        </linearGradient>
        <mask id="a" width="186" height="20">
            <rect width="186" height="20" rx="3"
            fill="#fff"/>
        </mask>
        <g mask="url(#a)">
            <path fill="#555" d="M0 0h31v20H0z" />
            <path fill="#007ec6"
            d="M31 0h155v20H31z"
            />
            <path fill="url(#b)" d="M0 0h186v20H0z" />
        </g>
        <g fill="#fff" text-anchor="middle" font-family="DejaVu Sans,
        Verdana,Geneva,sans-serif" font-size="11">
            <text x="16" y="15" fill="#010101"
            fill-opacity=".3">
                DOI
            </text>
            <text x="16" y="14">
                DOI
            </text>
            <text x="108"
            y="15" fill="#010101" fill-opacity=".3">
                10.5281/zenodo.4498634
            </text>
            <text x="108" y="14">
                10.5281/zenodo.4498634
            </text>
        </g>
    </svg>
```

jmwright, 2021-04-28:
It's taking about that long for the image to load for me in the US too. Maybe GitHub is timing out while trying to load the image? 

I tried uploading the SVG to my own web server and it loads fine in the readme. I guess I'll change the readme to point to the temporary image location and leave this issue open so we can circle back when Zenodo has their issue fixed.

jmwright, 2021-04-28:
Ok, change made and badge is visible again. Thanks for the feedback on this issue @fedorkotov

jmwright, 2021-04-29:
Looks like there might be a content header issue on Zenodo's side in addition to the long load times.

https://github.com/zenodo/zenodo/issues/2165#issuecomment-829109421

jmwright, 2021-05-04:
This appears to be fixed now on Zenodo's side, although there has been no communication on what happened. The issue was closed and the badge SVG now works.

closed 2021-05-04 by jmwright

adam-urbanczyk, 2021-05-04:
Thanks for looking into this @jmwright !

PR #748 (2021-04-28): Fixed copy-paste error in docs
Also fixed the DOI badge to close #743