CadQuery/cadquery #746: [ENHANCEMENT] Add fillet and radius parameter to Workplane.rect() method

Jojain, 2021-04-24:
Hello, 

I found myself needing to have a rounded rectangle to loft to, and realized that rect doesn't have, fillet or chamfer parameters.
Since offset2D and chamfer2D have been implemented some time ago I think supporting this could be nice.

Looking at the source code for `rect()` it doesn't seems too complicated to implement.

For reference here a quick plugin I made that do rounded rect :

```def rounded_rect(self, xlen, ylen, fillet_radius):
    rect = cq.Workplane().rect(xlen, ylen).val()
    pts = rect.Vertices()
    rect = rect.fillet2D(fillet_radius, pts)
    return self.eachpoint(lambda loc: rect.moved(loc), True)
cq.Workplane.rounded_rect = rounded_rect```

![image](https://user-images.githubusercontent.com/50221409/115973431-354d6580-a555-11eb-98c1-f7b7732e8d6b.png)

adam-urbanczyk, 2021-04-25:
In terms of API we want to go in a different direction - define a separate `cq.Sketch` class (similar to cq.Workplane but only for 2D) that will handle this kind of operations and will be integrated with `cq.Wokrplane`.

Jojain, 2021-04-25:
I see, is it then planned to move all the 2D geometry from `cq.Workplane` to `cq.Sketch` ? (i.e rect, circle, ellipse, etc.).
Is `cq.Sketch` planned soon ? Or it's a long term view ?

jmwright, 2021-04-25:
@Jojain No ETA on this feature yet. It will depend on when someone has both the time and motivation to tackle it. Here are a couple of related issues.

1. #42 
2. #52

fozbstudios, 2021-06-27:
so is this still a no go with the other feature with no ETA? The code seems to be right there. Even if it is a band-aid eventually rendered moot by sketch(), does it hurt anything?

adam-urbanczyk, 2021-06-27:
Pretty much yes, I'm working on sketching. In the meantime you can use the code above as a plugin.

closed 2021-06-27 by adam-urbanczyk