CadQuery/cadquery #751: Splitting with arbitrary plane

fedorkotov, 2021-04-29:
What is the proper way of splitting geometry with an arbitrary plane?

### Is `Workplane.split(..)` bugged or am I using it wrong?
[Documentation suggests](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=split#cadquery.Workplane.split) that `Workplane.split(..)` should accept `Shape` or `Workplane` 

Even though `Plane` is not a `Shape` split accepts it without error but splits the model along current XY plane instead,
```python
def get_body():
    return cq.Workplane("XY")\
    .rect(1,1)\
    .extrude(1)

result = \
    get_body()\
    .split(
        cq.Plane.named("ZX")
        .rotated((0,-30,0)))
```
[image]
(should produce wedge shape, see below)

If I create a `Workplane` from my plane and pass it to `split` it produces `ValueError: Null TopoDS_Shape object` exception
```python
def get_body():
    return cq.Workplane("XY")\
    .rect(1,1)\
    .extrude(1)

cutting_workplane = \
    cq.Workplane(
        inPlane = \
            cq.Plane.named("ZX")
            .rotated((0,-30,0)),
        origin = cq.Vector(0,0,0))

result = \
    get_body()\
    .split(cutting_workplane)
```

### Can I patch `Workplane.plane`?

The following code works and produces expected result but I looks ugly. 
Is this Ok thing to do? Is `Workplane.plane` an implementation detail or part of "public" interface of `Workplane`?

```python
def splitTop(wp, plane):
    old_plane = wp.plane
    wp.plane = plane
    wp = wp.split(keepTop=True)
    wp.plane = old_plane
    return wp

def get_body():
    return cq.Workplane("XY")\
    .rect(1,1)\
    .extrude(1)
    
result = splitTop(
    get_body(), 
    cq.Plane.named("ZX")\
    .rotated((0,-30,0)))
```
[image]

jmwright, 2021-04-29:
The following works for me.
```python
result = (cq.Workplane().box(10, 10, 10)
                        .faces(">Z").transformed(rotate=(45, 45, 45))
                        .split(keepBottom=True))
```
`transformed` creates a workplane based on the current one and applies rotation and/or translation to it.

[image]

fedorkotov, 2021-04-30:
@jmwright I know about that possibility but I want to specify the cutting plane in world coordinate system without knowledge of particular part geometry.  `transformed(..)` works fine only for symmetric parts centered in some way (`CenterOfMass`, `ProjectedOrigin`, `CenterOfBoundBox`) around world center of origin. I probably could have figured out offset of part center relative to world center of origin and pass it to `transformed(..)`. But at this point the whole process becomes even more complex than my `Workplane.plane` patching.

adam-urbanczyk, 2021-04-30:
You can construct a face and pass it to split. In the example above you did not pass any topo object, just an empty Workplane.

fedorkotov, 2021-04-30:
@adam-urbanczyk Thank you. This is closer to what I wanted to do. A lot of code to write for such a simple task but it works.

To create face big enough I have to know part size and location. Is there a way to get a bounding box of all objects in Workplane?

```python
from cadquery.selectors import DirectionMinMaxSelector

def get_body():
    return cq.Workplane("XY")\
    .rect(1,1)\
    .extrude(1)
    
result = \
    get_body()\
    .split(
        cq.Workplane()
        .add(cq.Face
               .makePlane(
                   5,5, 
                   cq.Vector(0,0,0.5), 
                   cq.Vector(0,-1,1))))\
    .solids(
        DirectionMinMaxSelector(
            cq.Vector(0,-1,1), 
            False))
```
[image]

marcus7070, 2021-04-30:
> [Documentation suggests](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=split#cadquery.Workplane.split) that `Workplane.split(..)` should accept `Shape` or `Workplane`
>
> Even [...]

The split method can take a boolean as it's first argument, so your `Plane` is being evaluated as `True` and your results are the same as `.split(keepTop=True)`.

https://github.com/CadQuery/cadquery/blob/2b7f39b234df15328ad0102b1b0c15bbd4f7efb4/cadquery/cq.py#L275-L352

> To create face big enough I have to know part size and location. Is there a way to get a bounding box of all objects in Workplane?

Like in the split code above, use `.findSolid()` to get a `Solid` or `Compound`, then get the `BoundBox` and use `DiagonalLength`. But you can actually make an infinite Face. The first two arguments to `Face.makeFace` can be `None`. This produces the same result as your code above:
```python
def get_body():
    return cq.Workplane("XY")\
    .rect(1,1)\
    .extrude(1)
    
result = (
    get_body()
    .split(cq.Face.makePlane(
        None,
        None,
        cq.Vector(0,0,0.5), 
        cq.Vector(0,-1,1))
    )
    .solids("<<Z")
)
```

fedorkotov, 2021-04-30:
> [..] But you can actually make an infinite Face
Thank you. Did not realize that. My problem is solved

Can I submit a pull request adding `Plane` support to `Workplane.split(..)`? I think that would be less confusing and easier to use.

marcus7070, 2021-04-30:
The OCCT kernel splits using Shapes, not Planes, and we like to follow the same patterns as the kernel wherever possible. @adam-urbanczyk, do you have an opinion here?

adam-urbanczyk, 2021-04-30:
No strong opinion, but I'd like to know what's confusing/lengthy about this code?
```python
 get_body().split(cq.Face.makePlane(None,None,cq.Vector(0,0,0.5), cq.Vector(0,-1,1)))
```

Then there also is: https://github.com/CadQuery/cadquery/issues/357

fedorkotov, 2021-04-30:
This is subjective of course. This is all my experience and opinion. This is what I find confusing and inconvenient

1. Requirement to use a face instead of plane. Without prior knowledge I assume that something called a face is a finite not necessarily flat 2D thing. And a plane is flat and infinite. This is why I did not investigate if face can be made infinite because it seems unnatural for me. Maybe I'm not alone in these assumptions.
2. Writing these `None, None` each time clutters the code
3. Lack of `Face` constructor variants for the usual `XY`, `ZX`, etc. planes and lack of `Face` method for rotation
    Compare this
    ```python
   cq.Plane.named("ZX").rotated((-30,0,0))
    ```
    to this
    ```python
    cq.Face.makePlane(None,None, cq.Vector(0,0,0), cq.Vector(0,-math.cos(math.pi/6), math.sin(math.pi/6)))
    ```

marcus7070, 2021-04-30:
After thinking about it a bit more, there is an opportunity to simplify `Workplane.split` here. 

* Keep the split on Shape through `Solid.split`/`BRepAlgoAPI_Splitter`, 
* write a bit of new code to handle splitting on a Plane (either `Workplane.plane` or an argument) by creating a Face (`.makePlane` method) and reusing the `Shape.split`, and
* drop all the `maxDim`, `topCutBox`, and the `Shape.cut` method of splitting.

Hang on, it's probably easier to write a draft PR than explain this further in a comment.

fedorkotov, 2021-05-01:
@marcus7070 Excellent. This is more or less how I would have implemented it.

lalebarde, 2023-04-23:
Where is _makePlane_ documented please ?

lorenzncode, 2023-04-23:
> Where is _makePlane_ documented please ?

You can find the signature here:

https://github.com/CadQuery/cadquery/blob/3b9ead11e67384b0e129888a14b541c0ad76325e/cadquery/occ_impl/shapes.py#L2389-L2395

The method does not currently have a docstring defined so is not included in the docs here:
https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Face

closed 2024-11-20 by adam-urbanczyk