CadQuery/cadquery #755: Splitting with face parallel to extrusion direction fails

fedorkotov, 2021-05-03:
I tried to apply the new to me method of splitting model with an infinite face (see #751) to the project I'm working on only to discover that the geometry produced by such a split is somehow broken. I can not cut into it with a wire produced by `offset2D` from one of it's faces. Minimum non-working example follows

```python
result = cq.Workplane("XY")\
  .rect(1,1)\
  .extrude(1)\
  .split(cq.Face.makePlane(
        None,
        None,
        cq.Vector(0,0,0),
        cq.Vector(0,1,0)))\
   .solids(">>Y")\
   .faces("<Y")\
   .wires()\
   .toPending()\
   .offset2D(-0.1)\
   .cutBlind(-0.1) 
```
fails with `ValueError: Null TopoDS_Shape object`

similar model split with a workplane works
```python
result = cq.Workplane("XY")\
  .rect(1,1)\
  .extrude(1)\
  .transformed(rotate=(90, 0,0))\
  .split(keepBottom=True)\
   .solids("<<Y")\
   .faces("<Y")\
   .wires()\
   .toPending()\
   .offset2D(-0.1)\
   .cutBlind(-0.1)
```
[image]

For some reason it fails only if I do the splitting with a face parallel to direction of extrusion. Same operations but with cutting face perpendicular to direction of extrusion works.
```python
result = cq.Workplane("XY")\
  .rect(1,1)\
  .extrude(1)\
  .split(cq.Face.makePlane(
        None,
        None,
        cq.Vector(0,0,0.5),
        cq.Vector(0,0,1)))\
   .solids("<<Z")\
   .faces(">Z")\
   .wires()\
   .toPending()\
   .offset2D(-0.1)\
   .cutBlind(-0.1)
```
[image]

marcus7070, 2021-05-03:
In this code:
```python
result = cq.Workplane("XY")\
  .rect(1,1)\
  .extrude(1)\
  .split(cq.Face.makePlane(
        None,
        None,
        cq.Vector(0,0,0),
        cq.Vector(0,1,0)))\
   .solids(">>Y")\
   .faces("<Y")\
   .wires()\
   .toPending()\
   .offset2D(-0.1)\
   .cutBlind(-0.1) 
```
You are creating a wire that lies in the XZ plane and extruding it in the -Z direction (perpendicular to your current workplane, the XY plane). 

Try this instead:
```python
result = cq.Workplane("XY")\
  .rect(1,1)\
  .extrude(1)\
  .split(cq.Face.makePlane(
        None,
        None,
        cq.Vector(0,0,0),
        cq.Vector(0,1,0)))\
   .tag("s")\
   .transformed(rotate=(90, 0, 0))\
   .solids(">>Y", tag="s")\
   .faces("<Y")\
   .wires()\
   .toPending()\
   .offset2D(-0.1)\
   .cutBlind(-0.1)
```

fedorkotov, 2021-05-04:
@marcus7070 Oh. I see. Thank you.

closed 2021-05-04 by fedorkotov