CadQuery/cadquery #756: Intersection of 2D objects

Andy-Valentine, 2021-05-05:
Hello,

as mentioned before I working on a specialized slicer for 3D printing and now I want to slice zero-volume objects (i.e. Faces) to create polylines (or open wires), which would represent my printhead movement. Now I created a sample Face:

```
v0 = cq.Vector(0, 0, 0)
v1 = cq.Vector(10, 5, 5)
v2 = cq.Vector(0, 0, 10)
v3 = cq.Vector(2, 10, 10)
v4 = cq.Vector(10, 7, 5)

e0 = cq.Edge.makeLine(v0, v1)
e1 = cq.Edge.makeLine(v2, v3)

curvedface=cq.Face.makeRuledSurface(e0,e1)
show_object(curvedface)
```
[image]

 which I would want to slice, for example, at height 2 by using **.section()**

`sect=curvedface.section(2)`

but **.section()** requires a solid object to work. Is there a way to convert my Face to Solid to use **.section()** ? or is there an equivalent of **.intersect()**  which could be applied to two Faces (one as shown above and the other one would be a generic flat XY plane) to yield a line?

Thanks for your help.

adam-urbanczyk, 2021-05-06:
I'd expect intersect to work on faces, but somehow I'm getting an empty face. You could go down to the kernel level:
```python
v0 = cq.Vector(0, 0, 0)
v1 = cq.Vector(10, 5, 5)
v2 = cq.Vector(0, 0, 10)
v3 = cq.Vector(2, 10, 10)
v4 = cq.Vector(10, 7, 5)

e0 = cq.Edge.makeLine(v0, v1)
e1 = cq.Edge.makeLine(v2, v3)

curvedface=cq.Face.makeRuledSurface(e0,e1)
show_object(curvedface)

f = cq.Face.makePlane(None, None, (0,2,5), (0,0,1))

import OCP

op = OCP.BRepAlgoAPI.BRepAlgoAPI_Common()
arg = OCP.TopTools.TopTools_ListOfShape()
tool = OCP.TopTools.TopTools_ListOfShape()

arg.Append(curvedface.wrapped)
tool.Append(f.wrapped)

op.SetArguments(arg)
op.SetTools(tool)
op.Build()

for e in op.SectionEdges():
    show_object(e)
```

bragostin, 2021-05-12:
I think I found a way, using ``BRepAlgoAPI_Section()`` instead of ``BRepAlgoAPI_Common()`` in the ``intersect`` function.
So in ``shapes.py``:
```python
from OCP.BRepAlgoAPI import BRepAlgoAPI_Section

    def intersect(self, *toIntersect: "Shape") -> "Shape":
        """
        Intersection of the positional arguments and this Shape.
        """

        #intersect_op = BRepAlgoAPI_Common()
        intersect_op = BRepAlgoAPI_Section()

        return self._bool_op((self,), toIntersect, intersect_op)
```
Then in ``cq.py``:
```python
    def findSolid(
        self, searchStack: bool = True, searchParents: bool = True
    ) -> Union[Face, Solid, Compound]:

        found = self._findType((Face, Solid, Compound), searchStack, searchParents)

        if found is None:
            message = "on the stack or " if searchStack else ""
            raise ValueError(
                "Cannot find a solid {}in the parent chain".format(message)
            )

        return found
```
After that this code:
```python
import cadquery as cq

v0 = cq.Vector(0, 0, 0)
v1 = cq.Vector(10, 5, 5)
v2 = cq.Vector(0, 0, 10)
v3 = cq.Vector(2, 10, 10)
v4 = cq.Vector(10, 7, 5)

e0 = cq.Edge.makeLine(v0, v1)
e1 = cq.Edge.makeLine(v2, v3)

curvedface = cq.Face.makeRuledSurface(e0,e1)
curvedface = cq.Workplane('XY').newObject([curvedface])

sect = curvedface.section(2.)
```
produces the desired result below:
[image]

adam-urbanczyk, 2021-05-13:
> Thanks. But how to turn this resulting curve back into a CQ-object? I'd like to apply some operations to this curve now, such as **positionAt()**, extract vertices etc.

`cq.Shape.cast`

bragostin, 2021-05-14:
@adam-urbanczyk playing with ``BRepAlgoAPI_Section()`` instead of ``BRepAlgoAPI_Common()`` I found the first to be more efficient (meaning the latter often crashes) on complex unioned shapes. Any insight on this?