CadQuery/cadquery #762: Generating text crashes after upgrading to CQ-editor 0.3 / CQ2.1

Andy-Valentine, 2021-05-11:
Basically, this line crashed my CQ-editor after I upgraded it:

`TEXT=cq.Workplane('YZ').text('Hello, World', 10, 0.3, font='Arial', halign='center', valign='center')`

Previously it was working fine (CQ-editor 0.1 + CQ 2.0)

Also, when I unmark this line, the rest of the code (1000+ lines) is working as previously, no problems.

jmwright, 2021-05-12:
Your title says CQ-editor 0.3, but I'm assuming you meant 0.2 since 0.3 hasn't been released yet?

Andy-Valentine, 2021-05-12:
[image]

I guess it installs 0.3 by default...

fedorkotov, 2021-05-12:
@jmwright this what is probably CQ-Editor installed from GitHub repo master branch with conda. I have this version installed too.

I can confirm that @Andy-Valentine's example causes both CQ-editor and standalone python interactive session to fail on Windows 8.1 without any error messages in console. 
I found this message in Windows Event Viewer (sorry for Russian - I have localized windows installed).
Basically it means that error was caused by some code in TKService.dll v7.5.1.0

```
Имя сбойного приложения: python.exe, версия: 3.9.4150.1013, метка времени: 0x6070876e
Имя сбойного модуля: TKService.dll, версия: 7.5.1.0, метка времени: 0x6080572e
Код исключения: 0xc0000005
Смещение ошибки: 0x000000000006e900
Идентификатор сбойного процесса: 0x2938
Время запуска сбойного приложения: 0x01d747157ae8f572
Путь сбойного приложения: C:\Users\KotovF\miniconda3\envs\cadquery\python.exe
Путь сбойного модуля: C:\Users\KotovF\miniconda3\envs\cadquery\Library\bin\TKService.dll
Идентификатор отчета: cb0dac24-b308-11eb-81ad-2c4d545628aa
Полное имя сбойного пакета: 
Код приложения, связанного со сбойным пакетом: 
```

marcus7070, 2021-05-12:
It's the `int`:

```
>>> import cadquery as cq
>>> cq.Workplane('YZ').text('Hello, World', 10.0, 0.3, font='Arial', halign='center', valign='center')
<cadquery.cq.Workplane object at 0x7f695442e4c0>
>>> cq.Workplane('YZ').text('Hello, World', 10, 0.3, font='Arial', halign='center', valign='center')
Segmentation fault (core dumped)
```

:exploding_head:

jmwright, 2021-05-14:
Can this be closed now that #764 has been merged?

closed 2021-05-14 by Andy-Valentine