CadQuery/cadquery #763: documentation: can a link to a community/forum be added?

GeorgesPaul, 2021-05-12:
1. Is there any good CADquery forum or community out there? 
2. If so: can it be added to readme.md? 

I'm running into issues where the manual doesn't seem extensive enough to figure out how to accomplish something and where I just can't figure out whether what I want is possible (and how). A community where CQ users can help each other out would be a great resource. It could be a reddit, forum, stackoverflow kind of place or maybe Github has such a feature?

Jojain, 2021-05-12:
There is a discord channel and a google group where you can ask for help :

Google group  : https://groups.google.com/g/cadquery
Discord : https://discord.gg/6R6RNAz9Zh

fedorkotov, 2021-05-12:
You can also ask for help or documentation enhancement here, at GitHub. Or maybe even find an answer in one of previously created issues. This is how I got all the help I needed for my project.

jmwright, 2021-05-12:
Here is the getting help documentation: https://github.com/CadQuery/cadquery#getting-help

jmwright, 2021-05-12:
@GeorgesPaul It looks like your original requests are already covered so I'm closing this. Please feel free to ask a question specific to the problem you're having in a separate issue here, or post to Discord/Google Groups.

closed 2021-05-12 by jmwright