CadQuery/cadquery #766: Subtracting hundreds of square holes (net)

infosisio, 2021-05-13:
Coming from openscad, I was able to create a grid net with hundreds of square holes quite easily.

As there are several ways to subtract pieces in cadquery, I would like to know which one is more suitable.

Something like this:
[image]

Jojain, 2021-05-13:
I think you could do it like this :
```python
import cadquery as cq
result = cq.Workplane("XY" ).box(50, 50, 0.5, centered = False)

def mesh_grid(x,y,N):
    pts = []
    for i in range(1,N):   
        for j in range(1,N):
            pts.append((i*x/N, j*y/N))
    return pts
pts = mesh_grid(50,50, 10)
result = result.pushPoints(pts).rect(1,1).cutThruAll()
```
Which gives the following result :

[image]

infosisio, 2021-05-13:
Oh so traditional iteration, fair enough, thank you!

Jojain, 2021-05-13:
Well that seems the most obvious to me. You would probably prefer to extrude your grid cross section directly with holes rather than cutting them because with high number of holes the cutting process takes really long.
```python
import cadquery as cq
result = cq.Workplane("XY" ).rect(50,50, centered =False)

def mesh_grid(x,y,N):
    pts = []
    for i in range(1,N):   
        for j in range(1,N):
            pts.append((i*x/N, j*y/N))
    return pts
pts = mesh_grid(50,50, 30)
result = result.pushPoints(pts).rect(1,1).extrude(2)
```

jmwright, 2021-05-13:
@infosisio As @Jojain alluded to, you can draw the outer rectangle with all of the inner rectangles and then extrude them together. CadQuery should figure out that the inner rects should be holes. I'm guessing that will be the fastest method for making the grid.

infosisio, 2021-05-13:
@jmwright Are you referring to the second example, exactly as it's written now? I am just wondering why should this take around 15 secs to render.

Jojain, 2021-05-13:
It indeed took a while when raising the N number. ( In the first example quickly took ages, in the second it run but it also gets quickly long with high N numbers)

infosisio, 2021-05-13:
I am just wondering, why, the tech is much better than openscad. In openscad I was able to use intersection and union and have it render way more holes instantly.

infosisio, 2021-05-14:
For reference, this was the openscad approach:

```
holeSize = 1;
strengthX = 0.2;
strengthY = 0.2;
strengthZ = 2;

x = 210;
x_steps = round(x/1.2);

y = 297;
y_steps = round(y/1.2);

module net(){
    intersection() {
        union() {
            for(x = [0:x_steps]) {
                translate([(holeSize + strengthX)*x, 0, 0])
                cube([strengthX, y, strengthZ]);
            }
            for(y = [0:y_steps]) {
                translate([0, (holeSize + strengthY)*y, 0])
                cube([x, strengthY, strengthZ]);
            }
        }  
    }
}
```

infosisio, 2021-05-14:
I was able to do this:

```
import cadquery as cq
result = cq.Workplane("XY")

def mesh_grid(x,y,N):
    pts_y = []
    pts_x = []    
    for i in range(2,23):   
        pts_y.append((0,i*y/20))
    for i in range(2,23):   
        pts_x.append((i*x/20+-30,30))        
    return pts_y, pts_x

pts_y, pts_x = mesh_grid(50,50,30)
result = result.pushPoints(pts_y).rect(50,1).extrude(2)
result = result.pushPoints(pts_x).rect(1,50).extrude(2)
```
Not perfect but much faster than 15 seconds.

marcus7070, 2021-05-14:
There is a native OCCT method for repeating shapes. I haven't benchmarked to see if it's any better than just creating all the wires first then extruding.

```python
import cadquery as cq
import OCP

def repeat(self, x=None, y=None, z=None, run_parallel=False):
    """
    Make this shape periodic.

    TODO: auto period from self.BoundingBox

    :param x: Parameters for the periodic repetition along the x axis. A two
      tuple, the first value is the period and the second is the number of
      repetitions (excluding the base Shape).
    :param y: Parameters for the periodic repetition along the y axis. Same two
      tuple as the x axis.
    :param z: Parameters for the periodic repetition along the z axis. Same two
      tuple as the x axis.
    :param run_parallel: Should the make parallel process be run in multiple threads?
    """

    mp = OCP.BOPAlgo.BOPAlgo_MakePeriodic()
    mp.SetShape(self.wrapped)
    if x is not None:
        mp.MakeXPeriodic(True, x[0])
    if y is not None:
        mp.MakeYPeriodic(True, y[0])
    if z is not None:
        mp.MakeZPeriodic(True, z[0])
    mp.SetRunParallel(run_parallel)
    mp.Perform()
    if x is not None:
        mp.XRepeat(x[1])
    if y is not None:
        mp.YRepeat(y[1])
    if z is not None:
        mp.ZRepeat(z[1])
    return cq.Shape.cast(mp.RepeatedShape())

cq.Shape.repeat = repeat

base_shape = (
    cq.Workplane()
    .rect(1, 1)
    .rect(1.1, 1.1)
    .extrude(2)
    .val()
)

num = 49
compound_grid = base_shape.repeat((1.1, num), (1.1, num), run_parallel=True)
solids = compound_grid.Solids()
first_solid = solids.pop(0)
grid = first_solid.fuse(*solids, glue=True).clean()

if "show_object" in locals():
    show_object(grid)
```

[image]

If you're going to try out the above code, I would reduce `num` to 5 to start with and work upwards.

> I am just wondering, why, the tech is much better than openscad.
> In openscad I was able to use union and have it render way more holes instantly.

I would say creating a series of boxes and subtracting them from a larger box is a perfect use case for a CSG kernel. If you want to see where a BREP kernel is better, then I would sweep some wires along a curve.

infosisio, 2021-05-14:
@marcus7070 

Unfortunately it is slower.

> There is a native OCCT method for repeating shapes. I haven't benchmarked to see if it's any better than just creating all the wires first then extruding.

Got it.

> I would say creating a series of boxes and subtracting them from a larger box is a perfect use case for a CSG kernel. If you want to see where a BREP kernel is better, then I would sweep some wires along a curve.

marcus7070, 2021-05-14:
Interesting, I just benchmarked my code above against the other method (of drawing all edges first then extruding):

```python
import cadquery as cq
from itertools import product

num = 49
hole_dim = 1
wall_thick = 0.2
outer_dim = num * hole_dim + (num + 1) * wall_thick

vals = [wall_thick + (hole_dim + wall_thick) * idx for idx in range(num)]
points = [(x, y) for x, y in product(vals, repeat=2)]

grid = (
    cq.Workplane()
    .rect(outer_dim, outer_dim, centered=False)
    .pushPoints(points)
    .rect(hole_dim, hole_dim, centered=False)
    .extrude(2)
)
```

```
> time python repeat.py
python repeat.py  33.97s user 1.46s system 122% cpu 29.008 total
> time python wires.py
python wires.py  111.78s user 0.09s system 99% cpu 1:51.89 total
```

Repeating the shape was 3x faster for me.

adam-urbanczyk, 2021-05-14:
@infosisio this is the current idiomatic way of constructing such structures:

```python
import cadquery as cq

N = 50
d = 2

res = (
    cq.Workplane()
    .rect(N*d+d, N*d+d)
    .rarray(d, d, N, N).rect(0.9*d, 0.9*d)
    .extrude(1, clean=False)
```

It is getting very slow above `N=20`, there is a known bottleneck in `cq.Face.makeFromWires`. 

Please stay with OpenSCAD unless you need a Python based solution or a B-Rep kernel.

Jojain, 2021-05-14:
I have tried it with CATIA V5, it's way better but still gets really really slow with high N number. I think it's clear that it's not a use case where brep kernel excels

infosisio, 2021-05-14:
> Interesting, I just benchmarked my code above against the other method (of drawing all edges first then extruding):
>
> ```python
> import cadquery as cq
> from itertools import product
>
> num = 49
> hole_dim [...]

Are we comparing exactly the version of Jojain and yours with the num set to 49 or 50? If so, yours loads at least 3 times slower for me, I am on the latest version of cq-editor.

> Please stay with OpenSCAD unless you need a Python based solution or a B-Rep kernel.

This is a particular use case, not so common thankfully. CQ is much better in every other way.

Thanks to all for the responses.

closed 2021-05-15 by infosisio