CadQuery/cadquery #767: Need to run Assemble.solve() twice.

slobberingant, 2021-05-14:
From google group discussion: https://groups.google.com/g/cadquery/c/LUeih7iqVrk
Assembly.solve() needs to be run twice otherwise the model creation fails.
Code is not minimal but the error is reproducible.

import cadquery as cq
from cadquery import exporters

class Trough():

    def __init__(self, width, length, height):
        self.width = width
        self.length = length
        self.height = height
        self.lip = 50
        self.thickness = 6
        self.assembly = cq.Assembly()

        self.add_base_to_assembly()
        self.add_sides_to_assembly()
        self.add_back_to_assembly()
        self.add_lips_to_assembly()
        self.assembly.solve() # run once and fail.
        self.assembly.solve() # run twice and all good.

    def make_base_plate(self):
        return cq.Workplane().box(
            self.width + 2 * self.thickness + 2 * self.lip, 
            self.length, 
            self.thickness)

    def make_side_plate(self):
        return cq.Workplane().box(self.length, self.height, self.thickness)

    def make_back_plate(self):
        return cq.Workplane().box(
            self.width + 2 * self.lip + 2 * self.thickness, 
            self.height + self.lip, 
            self.thickness)

    def make_lip_plate(self):
        return cq.Workplane().box(self.length, self.lip, self.thickness)

    def add_base_to_assembly(self):
        self.assembly.add(self.make_base_plate(), name='base_plate')

    def add_sides_to_assembly(self):
        x = self.width / 2 - self.thickness
        translations = (
            ((x, 0, 0), 'side_plate_lhs', '>X'),
            ((-x, 0, 0), 'side_plate_rhs', '<X')
        )
        for coord, name, direction in translations:
            (
                self.assembly
                .add(self.make_side_plate(), name=name)
                .constrain('base_plate',
                           self.make_base_plate().faces('>Z').val().translate(coord), 
                           name, 
                           self.make_side_plate().faces('>Y').val(),
                           'Plane')
                .constrain('base_plate@faces@>Y', '{}@faces@{}'.format(name, direction), 'Axis')
            )
    
    def add_back_to_assembly(self):
        translation = (0, 0, self.height / 2 - self.lip / 2 + self.thickness/2)
        (
            self.assembly
            .add(self.make_back_plate(), name='back_plate')
            .constrain('base_plate',
                       self.make_base_plate().faces('>Y').val().translate(translation), 
                       'back_plate', 
                       self.make_back_plate().faces('>Z').val(),
                       'Plane')
            .constrain('base_plate@faces@>Z', 'back_plate@faces@>Y', 'Axis')
        )

    def add_lips_to_assembly(self):
        translation = (self.width/2, 0, self.height - self.thickness)
        (
            self.assembly
            .add(self.make_lip_plate(), name='lip_lhs')
            .constrain('base_plate@faces@>X', 'lip_lhs@faces@>Y', 'Axis')
            .constrain('base_plate',
                       self.make_base_plate().faces('>Z').val().translate(translation), 
                       'lip_lhs', 
                       self.make_lip_plate().faces('>Z').val(),
                       'Plane')
        )

width = 900
length = 1000
height = 300

mytrough = Trough(width, length, height)
if "show_object" in locals():
    show_object(mytrough.assembly)

adam-urbanczyk, 2021-05-14:
NB: `DIFF_EPS` seems to have impact on this.

slobberingant, 2021-05-23:
Is it possible that assemblies that are not constrained completely can confuse the solver and cause this kind of behavior?

adam-urbanczyk, 2021-05-23:
No, I don't think so.

adam-urbanczyk, 2021-06-21:
Additional follow-up

- [ ] Store the constraint solver status in e.g. `cq.Assembly._solver_result`
- [ ] Re-implement the jacobian using finite diff on f instead of f^2 (i.e. df^2 = 2*f*df)

adam-urbanczyk, 2021-06-27:
@slobberingant could you check if #806 solves the issue?

slobberingant, 2021-07-02:
I have tested with this code and another that had similar problems. It appears to be working well now. Thank you.

PR #806 (2021-07-03): Assy solver fix
Will resolve #767, #790, #809
closed 2021-07-02 by slobberingant