CadQuery/cadquery #769: Drawing T-Slot Front Face

infosisio, 2021-05-14:
I am sorry for the naive question, I've been able to draw big objects fine by connecting lines or polylines, but I seem to be getting something wrong about the scale of drawing something small/precise like the front face of a t slot profile.

[image]

Could you please let me know of what is the right approach for that?

fedorkotov, 2021-05-14:
> I've been able to draw big objects fine by connecting lines or polylines, but I seem to be getting something wrong about the scale of drawing something small/precise like the front face of a t slot profile

What is the difference between big and small when you are using CAD? Methods of manufacturing the thing may be quite different depending on material, required tolerances, size and quantity. But when you are creating a CAD model it does not matter. Or am I missing something?

This is one way you can draw the part on your picture in CadQuery
```python
result = cq.Workplane("YZ")\
    .line(1,0)\
    .line(1,1)\
    .line(0,1)\
    .line(-1,0)\
    .line(0,1)\
    .line(-1,0)\
    .mirrorY()\
    .extrude(1.5, both=True)\
    .faces(">Z")\
    .workplane()\
    .circle(0.8)\
    .extrude(1)\
    .faces(">Z")\
    .workplane()\
    .circle(0.5)\
    .cutBlind(-10)
```
[image]

infosisio, 2021-05-14:
Thanks a lot! I am fairly new to cadquery, I was using lineto before for objects similar to the examples listed.

closed 2021-05-14 by infosisio

fedorkotov, 2021-05-14:
Take a look at [examples](https://cadquery.readthedocs.io/en/latest/examples.html) page of documentation. [This one](https://cadquery.readthedocs.io/en/latest/examples.html#mirroring-symmetric-geometry) and [that one](https://cadquery.readthedocs.io/en/latest/examples.html#mirroring-3d-objects) in particular may be of interest.

> I am coming from openscad which works vastly differently from traditional cad.

CadQuery is both similar to OpenSCAD and different. Two main differences IMHO are
- Model is written in python which is universal language with access to vast collection of libraries unrelated to CadQuery for any task you can think of
- Underlying engine (Open Cascade) is based on [boundary representation](https://en.wikipedia.org/wiki/Boundary_representation) and is much more powerful than [Constructive Solid Geometry](https://en.wikipedia.org/wiki/Constructive_solid_geometry) based CGAL used by OpenSCAD.

infosisio, 2021-05-14:
@fedorkotov Thanks, I will take a look. When it comes to the design process of the two programs, there is no similarity. In openscad there is no concept of planes, lines etc, you are simply adding geometric primitives and use hull to create more complex shapes out of them.