CadQuery/cadquery #771: STL Import

infosisio, 2021-05-18:
I have the impression that this is not currently supported. If so, why not?

adam-urbanczyk, 2021-05-18:
Correct, only STL export is supported. This will not change in the foreseeable future. Converting an STL to B-Rep is rather involved.

closed 2021-05-18 by adam-urbanczyk

jpoles1, 2022-07-31:
I had this same question and came across this issue. Would be a nice feature.

I went looking and came across this [thread](https://groups.google.com/g/cadquery/c/Gg1PU2duXoA) which lead me to the following [library](https://github.com/JustinSDK/cqMore) + [script](https://github.com/JustinSDK/cqMore/blob/main/examples/import_stl.py). It's a bit slow, but seems to work ok for basic STLs (though it was giving me some errors with a more complicated one)