CadQuery/cadquery #777: Workplane.mirrorY duplicates pending wires

marcus7070, 2021-05-26:
Say you want to loft between two loops that are symetrical around the Y axis.

```python
import cadquery as cq

step0 = (
    cq.Workplane()
    .spline([(0, 0), (2, 2), (0, 3)])
    .mirrorY()
)

step1 = (
    step0
    .transformed(offset=(0, 0, 1))
    .spline([(0, 1.5), (1, 2), (0, 3)])
    .mirrorY()
)

part = step1.loft()
```

`step0` looks correct:
[image]

`step1` however has 3 loops in it, when I expect 2:
[image]

`mirrorY` acts on all pending wires. As a side note: `mirrorY` with out of plane wires reflects the wire about the current workplane. Hence my large loop is mirrored from below the plane to above it.

And my loft winds up looking like this:
[image]

I'm not really sure how to proceed here. Maybe I just need to accept that `mirrorY` will behave like this.

My intuition is that `mirrorY` should act only on edges and wires in the current plane, but we have no way to designate that wires "belong" in a certain plane.

Here is a bit of a radical idea: `CQContext.pendingWires`/`pendingEdges` could be changed to a dict with a key representing the current workplane (perhaps `id(self.Plane)`) and a list of pending Wires or Edges as the value? Then `mirrorX`/`Y`, `offset2D` etc. could just act on the wires created in the current workplane, and `loft` etc. could act on all values?

adam-urbanczyk, 2021-05-26:
I'd propose to not touch `pendingWires` semantics and wait for `cq.Sketch`. I want to start with it after #735 .

adam-urbanczyk, 2021-05-26:
The workaround is to construct the wires separately:
```python
import cadquery as cq

step0 = (
    cq.Workplane()
    .spline([(0, 0), (2, 2), (0, 3)])
    .mirrorY()
)

step1 = (
    cq.Workplane()
    .transformed(offset=(0, 0, 1))
    .spline([(0, 1.5), (1, 2), (0, 3)])
    .mirrorY()
)

part = cq.Workplane().add(step0).add(step1).toPending().loft()
```