CadQuery/cadquery #780: cadquery in pyqt5

Govert123, 2021-05-31:
Hi All,

I have been reading a lot about the cadquery project and I like it a lot. I would even like to use the library in my own program, which is written in python and uses various other libraries (matlib, qt, pyinstaller, mysql etc.).

With the code below, I can make a box in a qt window. 
    __self.canvas = qtDisplay.qtViewer3d(self)
    self.horizontalLayout.addWidget(self.canvas)
    self.display = self.canvas._display
    a_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()
    self.ais_box = self.display.DisplayShape(a_box)[0]
    self.display.FitAll()__

Ideally I would like to use a similar code to create the cadquery object and show that box. Like:
    _box = cadquery.Workplane("front").box(2.0, 2.0, 0.5)
    self.box2 = self.display.DisplayShape(box.shape())[0]_

I cannot find any examples to implement the CQ models or workplanes into the pyqt5 environment. 
Any help would be greatly appreciated.

jmwright, 2021-05-31:
@Govert123 CQ-editor is the desktop IDE that is widely used by the community and it's implemented in PyQT5.

https://github.com/CadQuery/CQ-editor

Govert123, 2021-05-31:
Hi Jmwright, 
Thank you for your quick response.
I have that example yes. The CQ editor is not a very straightforward program. I am struggling to find how it works. As this is not my native environment, I am struggling with the various contributions of OCC, OCT, AIS, CAF, VKT and CQ. I will update the comments when i find something.

jmwright, 2021-05-31:
I have this stashed in my notes from a much earlier conversation about a stand-alone viewer. I don't know if you'll find it helpful at all.
```
import sys
from OCP.BRepPrimAPI import BRepPrimAPI_MakeBox
from cq_editor.widgets import viewer as cqv
import cadquery as cq

app = cqv.QApplication(sys.argv)
viewer = cqv.OCCViewer()

dlg = cqv.QDialog()
dlg.setFixedHeight(400)
dlg.setFixedWidth(600)

cqv.layout(dlg,(viewer,),dlg)
dlg.show()

#box = BRepPrimAPI_MakeBox(20,20,30)
box = cq.Workplane("XY").box(20,20,30).val().wrapped
box_ais = cqv.AIS_ColoredShape(box)

viewer.display(box_ais)
sys.exit(app.exec_())
```
[This helper](https://github.com/CadQuery/CQ-editor/blob/3a87ed08c053afe74ad811938ba5979b7d3b37ff/cq_editor/cq_utils.py#L46) can also be used in conjunction with this.

Govert123, 2021-06-01:
Thank you for this push in this direction. I now have half the code from CQ as dependency in my program, but it works!

jmwright, 2021-06-17:
@Govert123 Feel free to reopen if you have other questions.

closed 2021-06-17 by jmwright