CadQuery/cadquery #781: Cadquery installation not working

BrandonEscamilla, 2021-06-01:
Hi all, 

I've been trying to install Cadquery in a docker container using conda environments unsuccesfully. I will attach my code here:  environment file and Dockerfile. I tried installing cadquery=2 cadquery=2.1 and cadquery=master, and no one is working in my installation.  I am not sure what I'm missing

**Dockerfile:**
```
FROM continuumio/miniconda3

WORKDIR /app

COPY . ./
# Create the environment:
RUN conda env create -f environment.yml

RUN chmod +x entry.sh

RUN echo "source activate env" > ~/.bashrc
ENV PATH /opt/conda/envs/env/bin:$PATH

EXPOSE 8000

ENTRYPOINT ["./entry.sh"]

```
**Environment.yml:**
```
name: env
channels:
  - cadquery
  - anaconda
  - conda-forge
  - defaults
dependencies:
  - pydantic=1.8.2
  - boto3=1.17.46
  - botocore=1.20.78
  - cadquery=master
  - python=3.8.10
  - python-dateutil=2.8.1
  - python-dotenv=0.17.1
  - python_abi=3.8
  - uvicorn=0.13.4
  - fastapi=0.65.1

```
**Error:**
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/conda/envs/env/lib/python3.8/site-packages/cadquery/__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/opt/conda/envs/env/lib/python3.8/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import (
ImportError: libGL.so.1: cannot open shared object file: No such file or directory
```

fedorkotov, 2021-06-01:
There are ready to use docker images. Have look at this section of readme https://github.com/CadQuery/cadquery#docker

adam-urbanczyk, 2021-06-01:
Also take a look here: https://github.com/ContinuumIO/docker-images/issues/49 or just install `libGL` in the image.

BrandonEscamilla, 2021-06-01:
I did follow the instructions and installing libgl worked, but now i am facing a new issue, do you have any idea what's the problem here?

`>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/conda/envs/env/lib/python3.8/site-packages/cadquery/__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/opt/conda/envs/env/lib/python3.8/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import (
ImportError: /opt/conda/envs/env/lib/python3.8/site-packages/OCP.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN18BRepOffset_Inter2d14ConnexIntByIntERK11TopoDS_FaceR17BRepOffset_OffsetR19NCollection_DataMapI12TopoDS_ShapeS6_23TopTools_ShapeMapHasherERKS8_RKN11opencascade6handleI14BRepAlgo_AsDesEEddRK18BRepOffset_AnalyseR22NCollection_IndexedMapIS6_S7_ER26NCollection_IndexedDataMapIS6_16NCollection_ListIS6_ES7_E`

adam-urbanczyk, 2021-06-01:
The missing symbol comes from OCCT - most likely you somehow end up with a wrong version. Can you list which packages were installed?

RubenRubens, 2021-06-04:
CadQuery Dockerfile: https://github.com/RubenRubens/cq-containers/blob/master/cq-conda/Dockerfile

Were you looking for this?

BrandonEscamilla, 2021-06-05:
Thank you all! now it's working. I still have some weird issues, like .text() not working in docker but working in my local. If other folks are having the same issues I faced before,  I will attach my running docker file, environment.yml and entry.sh. This project is just to setup fast api and import cadquery within a route.

entry.sh
```
#!/bin/sh
exec uvicorn app:app --host 0.0.0.0 --port 8000

```

Dockerfile
```

FROM continuumio/miniconda3

RUN apt-get update -y && \
	apt install -y libgl1-mesa-glx && \
	apt-get clean && \
	rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*

WORKDIR /app

COPY . ./
# Create the environment:
RUN conda env create -f environment.yml

RUN chmod +x entry.sh

RUN echo "source activate env" > ~/.bashrc
ENV PATH /opt/conda/envs/env/bin:$PATH
    
EXPOSE 8000 

ENTRYPOINT ["./entry.sh"]
```

environment.yml
```
name: env
channels:
  - cadquery
  - anaconda
  - conda-forge
  - defaults
dependencies:
  - pydantic=1.8.2
  - boto3=1.17.46
  - botocore=1.20.78
  - cadquery=2
  - python=3.8
  - uvicorn=0.13.4
  - fastapi=0.65.1
  - requests
```

marcus7070, 2021-06-05:
On Sat, 5 Jun 2021, 9:40 am Brandon Escamilla, ***@***.***>
wrote:

> I still have some weird issues, like .text() not working in docker but
> working in my local.
>
You probably don't have any fonts installed in your docker image?

>

BrandonEscamilla, 2021-06-05:
@marcus7070 I didn't even know I need to manually install the fonts, how can I do that? 😮

jmwright, 2021-06-05:
Does `/usr/share/fonts` exist and does it have fonts in it (i.e. ttf files)? You should be able to run `fc-cache -f -v` to make the system aware of any new font files that you put in there.

You can also specify a path to the font file for the text operation [like this](https://github.com/CadQuery/cadquery/blob/master/tests/test_cadquery.py#L3285). You could download the [ttf file from the tests](https://github.com/cactrot/cadquery/blob/a947d0b4be17c76dba0c25c79352bd8e6f572b26/tests/testdata/OpenSans-Regular.ttf) and specify the path to it.

raphaelsetin, 2021-06-11:
I am having the same problem as @BrandonEscamilla described, even though everything was building and working successfully before. Let me explain...

Please take a look on this if possible @adam-urbanczyk @jmwright 

# The problem

My last successful build was around 06/01/2021, and I got it by following the original installation guide with Conda, inside a Docker container (you will see the Dockerfile in a bit). However, recently, my builds succeed but a weird error appears when I try to import CADQuery:

**Error in shell:**
```
{"errorMessage": "Unable to import module 'Main': /opt/python/lib/python3.8/site-packages/OCP.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN18BRepOffset_Inter2d14ConnexIntByIntERK11TopoDS_FaceR17BRepOffset_OffsetR19NCollection_DataMapI12TopoDS_ShapeS6_23TopTools_ShapeMapHasherERKS8_RKN11opencascade6handleI14BRepAlgo_AsDesEEddRK18BRepOffset_AnalyseR22NCollection_IndexedMapIS6_S7_ER26NCollection_IndexedDataMapIS6_16NCollection_ListIS6_ES7_E", "errorType": "Runtime.ImportModuleError", "stackTrace": []}
```

**Error in Docker:**
```
START RequestId: 98582f22-c58c-41b8-80b2-6380a8d3a670 Version: $LATEST

Traceback (most recent call last): Unable to import module 'Main': /opt/python/lib/python3.8/site-packages/OCP.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN18BRepOffset_Inter2d14ConnexIntByIntERK11TopoDS_FaceR17BRepOffset_OffsetR19NCollection_DataMapI12TopoDS_ShapeS6_23TopTools_ShapeMapHasherERKS8_RKN11opencascade6handleI14BRepAlgo_AsDesEEddRK18BRepOffset_AnalyseR22NCollection_IndexedMapIS6_S7_ER26NCollection_IndexedDataMapIS6_16NCollection_ListIS6_ES7_E

END RequestId: 98582f22-c58c-41b8-80b2-6380a8d3a670

REPORT RequestId: 98582f22-c58c-41b8-80b2-6380a8d3a670 Init Duration: 0.26 ms Duration: 682.23 ms Billed Duration: 700 ms Memory Size: 5120 MB Max Memory Used: 5120 MB
```

PS: My `Main.lambda_handler` is basically just doing `import cadquery as cq`, and the problem is not in there.

# Hypothesis

I can see that `OCP 7.5beta` is being installed for today's `cadquery=master` version, and I don't think that same OCP was being installed before for the master version prior to 06/01/2021.

In addition, when `conda install -y -c conda-forge -c cadquery cadquery=master` runs, I keep getting weird messages in the shell about like:
```
#7 [4/7] RUN source activate cad &&     conda install -y -c conda-forge -c cadquery cadquery=master
#7 sha256:f00af22aa10f36ee1b5b7f5a883e2b0244b5b7f31e3d9ea85db4e40b80fa2037
#7 1.050 Collecting package metadata (current_repodata.json): ...working... done
#7 11.88 Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.
#7 22.10 Solving environment: ...working... failed with repodata from current_repodata.json, will retry with next repodata source.
#7 32.48 Collecting package metadata (repodata.json): ...working... done
#7 76.31 Solving environment: ...working...
#7 100.5 Warning: 2 possible package resolutions (only showing differing packages):
#7 100.5   - cadquery/linux-64::ocp-7.5beta-1_py3.8, conda-forge/linux-64::occt-7.5.2-h7391655_0
#7 100.5   - cadquery/linux-64::ocp-7.5.1beta-1_py3.8, conda-forge/linux-64::occt-7.5.1-h7391655_2done
#7 101.1
#7 101.1 ## Package Plan ##
#7 101.1
#7 101.1   environment location: /miniconda/envs/cad
#7 101.1
#7 101.1   added / updated specs:
#7 101.1     - cadquery=master
#7 101.1
#7 101.1
#7 101.1 The following packages will be downloaded:
```

# Additional Info

Python version: `v3.8`
My striped Dockerfile is:
```
ARG PYTHON_VERSION="3.8"
FROM public.ecr.aws/lambda/python:$PYTHON_VERSION AS build-image-2
ARG PYTHON_VERSION

# Arguments
ARG CONDA_ENV="cad"
ARG FUNCTION_DIR="Function Code"

# Set the user
USER root

[... 2312 lines cut ...]
#5 28.95  85350K .......... .......... .......... .......... .......... 92% 22.0M 0s
#5 28.96  85400K .......... .......... .......... .......... .......... 92% 15.7M 0s
#5 28.96  85450K .......... .......... .......... .......... .......... 92% 16.2M 0s
#5 28.96  85500K .......... .......... .......... .......... .......... 92% 20.5M 0s
#5 28.97  85550K .......... .......... .......... .......... .......... 93% 9.70M 0s
#5 28.97  85600K .......... .......... .......... .......... .......... 93% 18.2M 0s
#5 28.97  85650K .......... .......... .......... .......... .......... 93% 22.4M 0s
#5 28.97  85700K .......... .......... .......... .......... .......... 93% 14.9M 0s
#5 28.98  85750K .......... .......... .......... .......... .......... 93% 17.8M 0s
#5 28.98  85800K .......... .......... .......... .......... .......... 93% 14.2M 0s
#5 28.98  85850K .......... .......... .......... .......... .......... 93% 14.8M 0s
#5 28.99  85900K .......... .......... .......... .......... .......... 93% 29.3M 0s
#5 28.99  85950K .......... .......... .......... .......... .......... 93% 11.9M 0s
#5 28.99  86000K .......... .......... .......... .......... .......... 93% 13.0M 0s
#5 29.00  86050K .......... .......... .......... .......... .......... 93% 15.7M 0s
#5 29.00  86100K .......... .......... .......... .......... .......... 93% 12.7M 0s
#5 29.00  86150K .......... .......... .......... .......... .......... 93% 17.8M 0s
#5 29.01  86200K .......... .......... .......... .......... .......... 93% 20.8M 0s
#5 29.01  86250K .......... .......... .......... .......... .......... 93% 16.7M 0s
#5 29.01  86300K .......... .......... .......... .......... .......... 93% 13.5M 0s
#5 29.02  86350K .......... .......... .......... .......... .......... 93% 14.4M 0s
#5 29.02  86400K .......... .......... .......... .......... .......... 93% 19.2M 0s
#5 29.02  86450K .......... .......... .......... .......... .......... 93% 16.0M 0s
#5 29.02  86500K .......... .......... .......... .......... .......... 94% 24.0M 0s
#5 29.03  865

adam-urbanczyk, 2021-06-29:
@raphaelsetin what you described is a different issue (not libGL related).
@BrandonEscamilla is your issue solved (mmissing libGL)?

raphaelsetin, 2021-06-29:
@adam-urbanczyk You're totally right. I ended up posting it as another issue (#790), which has already been cleared out already from what we have discussed.

BrandonEscamilla, 2021-07-06:
Hey @adam-urbanczyk my issue is solved now. In my latest comment I added the config that worked for me! Thank you!

closed 2021-07-06 by BrandonEscamilla