CadQuery/cadquery #784: Load Assembly From STEP

infosisio, 2021-06-07:
I am able to create an assembly in CQ from scratch and export that to an STEP. So far so good.

How do I deal with the problem though where I am given an STEP assembly from someone and I want to handpick solids from it, in order to compose a new assembly or do further work?

Is there support for something like this?

adam-urbanczyk, 2021-06-07:
Currently STEP is imported as a `cq.Workplane` - you can select solids using CQ selectors.

infosisio, 2021-06-07:
@adam-urbanczyk Ok, thanks. Is there a quick example you could share for that?

jmwright, 2021-06-17:
It's not documented and not used a lot (as far as I know), but here's a simple example using the door assembly in the docs.

```python
import cadquery as cq

door = cq.importers.importStep('door_assy.step')

bottom = door.solids("<Z")

corner2 = door.solids(cq.selectors.NearestToPointSelector((0, 0, 0)))

show_object(corner2)
```
Not all selectors will work with solids, and some don't work in an intuitive way.

closed 2021-07-17 by adam-urbanczyk