CadQuery/cadquery #785: Workplane center

parnold-x, 2021-06-09:
Hi, if I try the example of center in the method documentation
s = cq.Workplane().box(1,1,1).faces(">Z")
s=s.center(-0.5,-0.5)
s=s.circle(0.25).extrude(0.2)

I get this result
[image]

I expected that the new origin of the workplane would be the top left corner. Am I missing something?

Jojain, 2021-06-09:
Why would you think the center would be in the top left corner ? The workplane center is, after the box operation, in (0,0,0)  you moved it in (-0.5,-0.5,0) so it makes sense to get your result

parnold-x, 2021-06-09:
Ok, thank you. Then it works.

closed 2021-06-09 by parnold-x

adam-urbanczyk, 2021-06-09:
You probably wanted:
```python
s = cq.Workplane().box(1,1,1).faces(">Z").workplane()
s=s.center(-0.5,-0.5)
s=s.circle(0.25).extrude(0.2)
```