CadQuery/cadquery #786: Asymmetric Chamfer Error

gumyr, 2021-06-09:
Asymmetric chamfer is not applied consistently.  The chamfers on the top of this ring look correct; however, the inner chamfer on the bottom is reversed.
[image]

Error found on cadquery master.
```python
import cadquery as cq

chamferBody = (
    cq.Workplane("XY")
    .circle(8)
    .circle(4)
    .extrude(4)
    .faces(">Z").chamfer(0.5,1.5)
    .faces("<Z").chamfer(0.5,1.5)
)
if "show_object" in locals():
    show_object(chamferBody,name="chamferBody")
```

adam-urbanczyk, 2021-06-09:
The order of the asymmetric chamfer is not well defined in general (I think that it depends on the orientation of the underlying wires).

gumyr, 2021-06-09:
I wondered if the cause of this issue was due to the difference in direction of the circular wires of the top and bottom of the doughnut hole. Is there anyway to make the result more predictable?

adam-urbanczyk, 2021-06-10:
Not that I know at the moment.

ksuszka, 2023-03-25:
I think I encountered the same issue on other simple example.

```python
    cq.Workplane()
    .box(10,10,10)
    .faces("<Z")
    .workplane()
    .rect(20,1)
    .cutThruAll()
    .faces("<Z")
    .chamfer(1.5,0.5)
```
[image]

kolod, 2024-04-28:
One more example:
```python
from cadquery import Workplane

for i, query in enumerate(["<X", ">X", "<Y", ">Y", "<Z", ">Z"]):
    result = Workplane(origin=(i*15.0,0,0)).box(10,10,10).faces(query).chamfer(2,1)
    show_object(result)
```
[image]

kolod, 2024-04-28:
I think the problem in edge normals.
```python
from cadquery import Workplane
from typing import List, cast
from cadquery.occ_impl.shapes import Edge

c1 = Workplane().box(10, 10, 10).faces("<Y")
for x in cast(List[Edge], c1.faces("<Y").edges().vals()):
    print(x.normal())
```

```plain
Vector: (-0.0, 1.0, 0.0)
Vector: (1.0, 0.0, -0.0)
Vector: (-0.0, 1.0, 0.0)
Vector: (1.0, 0.0, -0.0)
```
And if add `wire()` nothing changes.
```python
c2 = Workplane().box(10, 10, 10).faces("<Y")
for x in cast(List[Edge], c2.faces("<Y").edges().wire().vals()):
    print(x.normal())
```