CadQuery/cadquery #787: Matrix Class rotateX, rotateY and rotateZ methods assume the provide angle is in radians

gumyr, 2021-06-09:
The rotateX, rotateY and rotateZ methods of the Matrix class assume the angle of rotation is in radians not degrees like all of the other cadquery methods.
```python
import cadquery as cq
import math

print("Unit X Vector rotated about the Z axis by 30 degrees:(",math.cos(math.radians(30)),",",math.sin(math.radians(30)),",0)")
unitXVector = cq.Vector(1,0,0)
m0 = cq.Matrix()
m0.rotateZ(30)
rotatedUnitXVector = m0.multiply(unitXVector)
print("Unit X Vector rotated by rotateZ(30):",rotatedUnitXVector)

m1 = cq.Matrix()
m1.rotateZ(math.radians(30))
rotatedUnitXVector = m1.multiply(unitXVector)
print("Unit X Vector rotated by rotateZ(math.radians(30)):",rotatedUnitXVector)
```
There is one line fix:
```python
    def _rotate(self, direction: gp_Ax1, angle: float):

        new = gp_Trsf()
        # new.SetRotation(direction, angle)
        new.SetRotation(direction, math.radians(angle))

        self.wrapped = self.wrapped * gp_GTrsf(new)
```
However, this will break backwards compatibility.

adam-urbanczyk, 2021-06-09:
Why do you actually want to use the Matrix class? Unless you want to do some low level things (extend CQ) I'd advise against it.

gumyr, 2021-06-09:
My immediate need was the creation of rotation functions for the cq.Vector class. However, the affine transformation matrix is an incredibly powerful component of linear algebra allowing for combinations of rotation, translations, skew, etc. Who knows what clever things users might do with it in the future?

adam-urbanczyk, 2021-06-10:
Especially with things like skew your quickly run into kernel issues.