CadQuery/cadquery #790: Undefined Symbol in OCP.cpython-38-x86_64-linux-gnu.so: _ZN18BRepOffset_Inter2d14ConnexIntByIntERK11TopoDS_Face....

raphaelsetin, 2021-06-15:
I am having the same problem as @BrandonEscamilla described **later** in his issue (#781), but since our main issues are different, then I am creating this post so that someone is able to give me proper help; and others can easily spot this error too.

# The problem

My last successful build was around 06/01/2021, and I got it by following the original installation guide with Conda, inside a Docker container (you will see the Dockerfile in a bit). However, recently, my builds succeed but a weird error appears when I try to import CADQuery in the built container:

**Error in shell:**
```
{"errorMessage": "Unable to import module 'Main': /opt/python/lib/python3.8/site-packages/OCP.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN18BRepOffset_Inter2d14ConnexIntByIntERK11TopoDS_FaceR17BRepOffset_OffsetR19NCollection_DataMapI12TopoDS_ShapeS6_23TopTools_ShapeMapHasherERKS8_RKN11opencascade6handleI14BRepAlgo_AsDesEEddRK18BRepOffset_AnalyseR22NCollection_IndexedMapIS6_S7_ER26NCollection_IndexedDataMapIS6_16NCollection_ListIS6_ES7_E", "errorType": "Runtime.ImportModuleError", "stackTrace": []}
```

**Error in Docker:**
```
START RequestId: 98582f22-c58c-41b8-80b2-6380a8d3a670 Version: $LATEST

Traceback (most recent call last): Unable to import module 'Main': /opt/python/lib/python3.8/site-packages/OCP.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN18BRepOffset_Inter2d14ConnexIntByIntERK11TopoDS_FaceR17BRepOffset_OffsetR19NCollection_DataMapI12TopoDS_ShapeS6_23TopTools_ShapeMapHasherERKS8_RKN11opencascade6handleI14BRepAlgo_AsDesEEddRK18BRepOffset_AnalyseR22NCollection_IndexedMapIS6_S7_ER26NCollection_IndexedDataMapIS6_16NCollection_ListIS6_ES7_E

END RequestId: 98582f22-c58c-41b8-80b2-6380a8d3a670

REPORT RequestId: 98582f22-c58c-41b8-80b2-6380a8d3a670 Init Duration: 0.26 ms Duration: 682.23 ms Billed Duration: 700 ms Memory Size: 5120 MB Max Memory Used: 5120 MB
```

**Everything** was building and working successfully before 06/01/2021, so definitely an update either in CADQuery or its dependencies must have broken the Conda `cadquery=master` version.

PS: My `Main.lambda_handler` is basically just doing `import cadquery as cq`, so the problem is not related to other things.

# Hypothesis

I can see that `OCP 7.5beta` is being installed for today's `cadquery=master` version, and I don't think that same OCP was being installed before for the master version prior to 06/01/2021. Even when using older versions, such as `cadquery=merge` and `cadquery=2.1`, I keep getting errors -- with different versions, the errors are different though.

In addition, when `conda install -y -c conda-forge -c cadquery cadquery=master` runs, I keep getting weird messages in the shell about like:
```
#7 [4/7] RUN source activate cad &&     conda install -y -c conda-forge -c cadquery cadquery=master
#7 sha256:f00af22aa10f36ee1b5b7f5a883e2b0244b5b7f31e3d9ea85db4e40b80fa2037
#7 1.050 Collecting package metadata (current_repodata.json): ...working... done
#7 11.88 Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.
#7 22.10 Solving environment: ...working... failed with repodata from current_repodata.json, will retry with next repodata source.
#7 32.48 Collecting package metadata (repodata.json): ...working... done
#7 76.31 Solving environment: ...working...
#7 100.5 Warning: 2 possible package resolutions (only showing differing packages):
#7 100.5   - cadquery/linux-64::ocp-7.5beta-1_py3.8, conda-forge/linux-64::occt-7.5.2-h7391655_0
#7 100.5   - cadquery/linux-64::ocp-7.5.1beta-1_py3.8, conda-forge/linux-64::occt-7.5.1-h7391655_2done
#7 101.1
#7 101.1 ## Package Plan ##
#7 101.1
#7 101.1   environment location: /miniconda/envs/cad
#7 101.1
#7 101.1   added / updated specs:
#7 101.1     - cadquery=master
#7 101.1
#7 101.1
#7 101.1 The following packages will be downloaded:
```

# Additional Info

Python version: `v3.8`
My striped Dockerfile is:
```
ARG PYTHON_VERSION="3.8"
FROM public.ecr.aws/lambda/python:$PYTHON_VERSION AS build-image-2
ARG PYTHON_VERSION

# Arguments
ARG CONDA_ENV="cad"
ARG FUNCTION_DIR="Function Code"

# Set the user
USER root

[... 2307 lines cut ...]
#5 28.94  85100K .......... .......... .......... .......... .......... 92% 18.7M 0s
#5 28.94  85150K .......... .......... .......... .......... .......... 92% 12.5M 0s
#5 28.94  85200K .......... .......... .......... .......... .......... 92% 15.0M 0s
#5 28.95  85250K .......... .......... .......... .......... .......... 92% 20.7M 0s
#5 28.95  85300K .......... .......... .......... .......... .......... 92% 15.1M 0s
#5 28.95  85350K .......... .......... .......... .......... .......... 92% 22.0M 0s
#5 28.96  85400K .......... .......... .......... .......... .......... 92% 15.7M 0s
#5 28.96  85450K .......... .......... .......... .......... .......... 92% 16.2M 0s
#5 28.96  85500K .......... .......... .......... .......... .......... 92% 20.5M 0s
#5 28.97  85550K .......... .......... .......... .......... .......... 93% 9.70M 0s
#5 28.97  85600K .......... .......... .......... .......... .......... 93% 18.2M 0s
#5 28.97  85650K .......... .......... .......... .......... .......... 93% 22.4M 0s
#5 28.97  85700K .......... .......... .......... .......... .......... 93% 14.9M 0s
#5 28.98  85750K .......... .......... .......... .......... .......... 93% 17.8M 0s
#5 28.98  85800K .......... .......... .......... .......... .......... 93% 14.2M 0s
#5 28.98  85850K .......... .......... .......... .......... .......... 93% 14.8M 0s
#5 28.99  85900K .......... .......... .......... .......... .......... 93% 29.3M 0s
#5 28.99  85950K .......... .......... .......... .......... .......... 93% 11.9M 0s
#5 28.99  86000K .......... .......... .......... .......... .......... 93% 13.0M 0s
#5 29.00  86050K .......... .......... .......... .......... .......... 93% 15.7M 0s
#5 29.00  86100K .......... .......... .......... .......... .......... 93% 12.7M 0s
#5 29.00  86150K .......... .......... .......... .......... .......... 93% 17.8M 0s
#5 29.01  86200K .......... .......... .......... .......... .......... 93% 20.8M 0s
#5 29.01  86250K .......... .......... .......... .......... .......... 93% 16.7M 0s
#5 29.01  86300K ..

adam-urbanczyk, 2021-06-16:
Try installing occt 7.5.1 for now.

raphaelsetin, 2021-06-16:
Hey @adam-trhon, thank you for the feedback. **It worked!** 🎉 

What I had to do was to explicitly tell what version of OCCT I needed, like this (and specify `v7.5.1`):
```
RUN source activate $CONDA_ENV && \
	conda install -y -c conda-forge -c cadquery occt=7.5.1 cadquery=master
```

Btw, I wonder if this will eventually break as updates are pushed to cadquery master... what do you think about that?

**Lastly, why did this happen out of nowhere anyways? I never had to specify an OCCT version before; I thought cadquery master had the specific OCCT version bound to it in its environment config file.**

adam-trhon, 2021-06-16:
@raphaelsetin I belive you mean @adam-urbanczyk ;)

adam-urbanczyk, 2021-06-16:
Yes, we need to specify ocp 7.5.1 and not just 7.5 in the deps.

raphaelsetin, 2021-06-28:
Hey @adam-urbanczyk, would you know why the default `conda install -y -c conda-forge -c cadquery cadquery=master` is still not working even though you have approved and merged my PR?

I have just tested it today with the same above Dockerfile, and I got the same error again of:
```
Traceback (most recent call last): Unable to import module 'Main': /opt/python/lib/python3.8/site-packages/OCP.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN18BRepOffset_Inter2d14ConnexIntByIntERK11TopoDS_FaceR17BRepOffset_OffsetR19NCollection_DataMapI12TopoDS_ShapeS6_23TopTools_ShapeMapHasherERKS8_RKN11opencascade6handleI14BRepAlgo_AsDesEEddRK18BRepOffset_AnalyseR22NCollection_IndexedMapIS6_S7_ER26NCollection_IndexedDataMapIS6_16NCollection_ListIS6_ES7_E
```

What is strange to me is that in Anaconda's [website](https://anaconda.org/CadQuery/cadquery/files?version=master), there are like 2 sets of builds on different dates for the same `master` version/label. I was able to check in my installation output that only the latest one is being installed, but it still used the old environment file since it was clearly getting the following **OCP** version:
```
#6 107.1     ocp-7.5beta                |          1_py3.8        42.9 MB  cadquery
```

What are your thoughts?

adam-urbanczyk, 2021-06-28:
Well the PR did not fix the root cause (conda specs, not CI env). Will be solved by #806

raphaelsetin, 2021-06-28:
Thank you so much for the quick feedback! I will keep an eye on that issue and will test everything and close this issue once it's finished.

raphaelsetin, 2021-08-19:
Running `conda install -y -c conda-forge -c cadquery cadquery=master` works now! Thank you @adam-urbanczyk 🎉

closed 2021-08-19 by raphaelsetin