CadQuery/cadquery #793: Old working script now fails with: gp_Dir::Crossed() - result vector has zero norm

nealie, 2021-06-20:
I have just upgraded CaqQuery on MacOS to the latest master of about last week, and a script that used to work nicely now crashes.

The script is as follows:

```import cadquery as cq

length = 20
width = 16.5
thickness = 3
hinge_radius = thickness / 2
hinge_length = 21
slit_length = 12.5
slit_width = 2
bevel_angle = 20
catch_length = 4
catch_width = 1.35
catch_height = 0.8

foot = cq.Workplane("XY").box(length, width, thickness)

catch1 = (
    foot.faces("<Y").workplane()
    .move((length/2)-(catch_length/2), 0)
    .rect(catch_length, catch_width)
    .workplane(offset=catch_height)         # No longer centres the workplane on the rect above.
    .move((length/2)-(catch_length/2), 0)   # So move it to where it's expected.
    .rect(catch_length, 0.1)
    .loft(combine=False)
    .last()
)

catch2 = catch1.mirror(mirrorPlane="XZ")

foot = foot.union(catch1).union(catch2)

hinge = (
    cq.Workplane("XZ")
    .circle(hinge_radius)
    .extrude(hinge_length)
    .translate((length/2, (width/2)+((hinge_length-width)/2), 0))
)

foot = foot.union(hinge)

slit1 = (
    cq.Workplane("XY").box(slit_length, slit_width, thickness*2)
    .translate(((length/2)-(slit_length/2)+hinge_radius, width/4, 0))
)

slit2 = slit1.mirror(mirrorPlane="XZ")

foot = foot.cut(slit1)
foot = foot.cut(slit2)  # <<<- Crashes here!

bevel = (
    cq.Workplane("YZ")
    .box(width*2, thickness*2, thickness)
    .rotate((0, 0, 0), (0, 1, 0), -bevel_angle)
    .translate((-((length/2)+(thickness/2)-0.3), 0, 0))
)

foot = foot.cut(bevel)

if __name__ != "__main__":
    show_object(foot)
else:
    print("Exporting...")
    foot.val().exportStl("keyboard_foot.stl", 1e-3)
```
The crash is as follows:
```Traceback (most recent call last):
  File "/Users/nealie/Documents/CAD/CadQuery/keyboard_foot.py", line 49, in <module>
    foot = foot.cut(slit2)
  File "/usr/local/Caskroom/miniconda/base/envs/cadquery/lib/python3.9/site-packages/cadquery/cq.py", line 3294, in cut
    newS = newS.clean()
  File "/usr/local/Caskroom/miniconda/base/envs/cadquery/lib/python3.9/site-packages/cadquery/occ_impl/shapes.py", line 380, in clean
    upgrader.Build()
OCP.Standard.Standard_Failure: gp_Dir::Crossed() - result vector has zero norm
```

I've tried moving around the cuts and it makes no difference what order they're in. It always crashes at the second cut, but if I remove the second cut, the next cut with the bevel works fine, so it seems to be a problem with the cylindrical hinge being cut twice.

For full information, the environment is as follows:
MacOS: 11.2.3
OCP: 7.5.1beta
CadQuery: master of about a week ago.

As an aside: the workplane() call used to create a workplane above the last feature, but now seems to be centred on the original object on the stack. This seems odd and means I had to add another move, but the documentation does now say this. It was just a surprise as my old working script made a very strange shape.

adam-urbanczyk, 2021-06-20:
Use `clean=False` as a workaround:
```python
import cadquery as cq

length = 20
width = 16.5
thickness = 3
hinge_radius = thickness / 2
hinge_length = 21
slit_length = 12.5
slit_width = 2
bevel_angle = 20
catch_length = 4
catch_width = 1.35
catch_height = 0.8

foot = cq.Workplane("XY").box(length, width, thickness)

catch1 = (
    foot.faces("<Y").workplane()
    .move((length/2)-(catch_length/2), 0)
    .rect(catch_length, catch_width)
    .workplane(offset=catch_height)         # No longer centres the workplane on the rect above.
    .move((length/2)-(catch_length/2), 0)   # So move it to where it's expected.
    .rect(catch_length, 0.1)
    .loft(combine=False)
    .last()
)

catch2 = catch1.mirror(mirrorPlane="XZ")

foot = foot.union(catch1).union(catch2)

hinge = (
    cq.Workplane("XZ")
    .circle(hinge_radius)
    .extrude(hinge_length)
    .translate((length/2, (width/2)+((hinge_length-width)/2), 0))
)

foot = foot.union(hinge)

slit1 = (
    cq.Workplane("XY").box(slit_length, slit_width, thickness*2)
    .translate(((length/2)-(slit_length/2)+hinge_radius, width/4, 0))
)

slit2 = slit1.mirror(mirrorPlane="XZ")

foot = foot.cut(slit1)
foot = foot.cut(slit2, clean=False)  # <<<- Crashes here!

bevel = (
    cq.Workplane("YZ")
    .box(width*2, thickness*2, thickness)
    .rotate((0, 0, 0), (0, 1, 0), -bevel_angle)
    .translate((-((length/2)+(thickness/2)-0.3), 0, 0))
)

foot = foot.cut(bevel, clean=False)
```

The default arg change for `workplane` is documented in `changes.md`.

nealie, 2021-06-21:
Thanks. This works.

So the question is: this is obviously a bug which we have to work around, so should I leave this pr open as an indicator of the problem, or is it something in OpenCascade itself, and therefore beyond the reach of CadQuery?

lorenzncode, 2022-09-22:
This is working for me with the occt7.6 branch.  Actually it is also working with master and ocp-7.5.3.0.  I can reproduce the error with master and ocp-7.5.1beta.

adam-urbanczyk, 2022-09-22:
Solved in OCCT/OCP 7.5.3

closed 2022-09-22 by adam-urbanczyk