CadQuery/cadquery #794: Assembly Color Component Values in Exported STEP File Are Shifted

jmwright, 2021-06-21:
If I run the following code I expect to get a STEP file that has an RGB for the box of `(0.5, 0.5, 0.5)`, but I see `(0.735356983052,0.735356983052,0.735356983052)` instead.
```python
import cadquery as cq

assy = cq.Assembly()

box = cq.Workplane().box(10, 10, 10)

assy.add(box, color=cq.Color(0.5, 0.5, 0.5, 1))

assy.save("/home/jwright/Downloads/color_test.step")
```
Here is the RGB line within the resulting STEP file:
```
#451 = COLOUR_RGB('',0.735356983052,0.735356983052,0.735356983052);
```
Changing the alpha value does not seem to make a difference, and I've tried two different versions of CadQuery that are from after the assembly addition. They both do the same thing.

The effect of this is that it causes validation errors when comparing CQ 2.x generated STEPs to the originals that have been checked by hand by the KiCAD team.

Another glitch that might be related is that the colors in the original KiCAD models are recognized by FreeCAD when the STEP is imported, but the colors in our exported STEP files are ignored. I don't know if this has something to do with the way layers are handled by the OCCT export process.

Here is the full output of the resulting STEP file.
```
ISO-10303-21;
HEADER;
FILE_DESCRIPTION(('Open CASCADE Model'),'2;1');
FILE_NAME('Open CASCADE Shape Model','2021-06-21T15:22:04',('Author'),(
    'Open CASCADE'),'Open CASCADE STEP processor 7.5','Open CASCADE 7.5'
  ,'Unknown');
FILE_SCHEMA(('AUTOMOTIVE_DESIGN { 1 0 10303 214 1 1 1 1 }'));
ENDSEC;
DATA;
#1 = APPLICATION_PROTOCOL_DEFINITION('international standard',
  'automotive_design',2000,#2);
#2 = APPLICATION_CONTEXT(
  'core data for automotive mechanical design processes');
#3 = SHAPE_DEFINITION_REPRESENTATION(#4,#10);
#4 = PRODUCT_DEFINITION_SHAPE('','',#5);
#5 = PRODUCT_DEFINITION('design','',#6,#9);
#6 = PRODUCT_DEFINITION_FORMATION('','',#7);
#7 = PRODUCT('CQ assembly','CQ assembly','',(#8));
#8 = PRODUCT_CONTEXT('',#2,'mechanical');
#9 = PRODUCT_DEFINITION_CONTEXT('part definition',#2,'design');
#10 = SHAPE_REPRESENTATION('',(#11,#15),#19);
#11 = AXIS2_PLACEMENT_3D('',#12,#13,#14);
#12 = CARTESIAN_POINT('',(0.,0.,0.));
#13 = DIRECTION('',(0.,0.,1.));
#14 = DIRECTION('',(1.,0.,-0.));
#15 = AXIS2_PLACEMENT_3D('',#16,#17,#18);
#16 = CARTESIAN_POINT('',(0.,0.,0.));
#17 = DIRECTION('',(0.,0.,1.));
#18 = DIRECTION('',(1.,0.,-0.));
#19 = ( GEOMETRIC_REPRESENTATION_CONTEXT(3) 
GLOBAL_UNCERTAINTY_ASSIGNED_CONTEXT((#23)) GLOBAL_UNIT_ASSIGNED_CONTEXT(
(#20,#21,#22)) REPRESENTATION_CONTEXT('Context #1',
  '3D Context with UNIT and UNCERTAINTY') );
#20 = ( LENGTH_UNIT() NAMED_UNIT(*) SI_UNIT(.MILLI.,.METRE.) );
#21 = ( NAMED_UNIT(*) PLANE_ANGLE_UNIT() SI_UNIT($,.RADIAN.) );
#22 = ( NAMED_UNIT(*) SI_UNIT($,.STERADIAN.) SOLID_ANGLE_UNIT() );
#23 = UNCERTAINTY_MEASURE_WITH_UNIT(LENGTH_MEASURE(1.E-07),#20,
  'distance_accuracy_value','confusion accuracy');
#24 = PRODUCT_RELATED_PRODUCT_CATEGORY('part',$,(#7));
#25 = SHAPE_DEFINITION_REPRESENTATION(#26,#32);
#26 = PRODUCT_DEFINITION_SHAPE('','',#27);
#27 = PRODUCT_DEFINITION('design','',#28,#31);
#28 = PRODUCT_DEFINITION_FORMATION('','',#29);
#29 = PRODUCT('f6464f20-d2c5-11eb-b9c5-54ee75e79fa7',
  'f6464f20-d2c5-11eb-b9c5-54ee75e79fa7','',(#30));
#30 = PRODUCT_CONTEXT('',#2,'mechanical');
[... 478 lines cut ...]
  );
#435 = NEXT_ASSEMBLY_USAGE_OCCURRENCE('7','','',#27,#67,$);
#436 = PRODUCT_RELATED_PRODUCT_CATEGORY('part',$,(#69));
#437 = CONTEXT_DEPENDENT_SHAPE_REPRESENTATION(#438,#440);
#438 = ( REPRESENTATION_RELATIONSHIP('','',#32,#10) 
REPRESENTATION_RELATIONSHIP_WITH_TRANSFORMATION(#439) 
SHAPE_REPRESENTATION_RELATIONSHIP() );
#439 = ITEM_DEFINED_TRANSFORMATION('','',#11,#15);
#440 = PRODUCT_DEFINITION_SHAPE('Placement','Placement of an item',#441
  );
#441 = NEXT_ASSEMBLY_USAGE_OCCURRENCE('8','','',#5,#27,$);
#442 = PRODUCT_RELATED_PRODUCT_CATEGORY('part',$,(#29));
#443 = MECHANICAL_DESIGN_GEOMETRIC_PRESENTATION_REPRESENTATION('',(#444)
  ,#413);
#444 = STYLED_ITEM('color',(#445),#83);
#445 = PRESENTATION_STYLE_ASSIGNMENT((#446));
#446 = SURFACE_STYLE_USAGE(.BOTH.,#447);
#447 = SURFACE_SIDE_STYLE('',(#448));
#448 = SURFACE_STYLE_FILL_AREA(#449);
#449 = FILL_AREA_STYLE('',(#450));
#450 = FILL_AREA_STYLE_COLOUR('',#451);
#451 = COLOUR_RGB('',0.735356983052,0.735356983052,0.735356983052);
ENDSEC;
END-ISO-10303-21;
```

adam-urbanczyk, 2021-06-22:
I tried importing an assy using FC 0.20 and I can see the colors. I checked the imported color value form the dark red object and it is [165,0,53] ~ [ 0.647, 0, 0.208] which is as expected (module the rounding). The alpha has indeed no effect (this is even mentioned in the docs AFAIR) - I was assuming (maybe wrongly) that it is not a part of the STEP spec.
```
import cadquery as cq

w = 10
d = 10
h = 10

part1 = cq.Workplane().box(2*w,2*d,h)
part2 = cq.Workplane().box(w,d,2*h)
part3 = cq.Workplane().box(w,d,3*h)

assy = (
    cq.Assembly(part1, loc=cq.Location(cq.Vector(-w,0,h/2)))
    .add(part2, loc=cq.Location(cq.Vector(1.5*w,-.5*d,h/2)), color=cq.Color(0.65,0,.21,1))
    .add(part3, loc=cq.Location(cq.Vector(-.5*w,-.5*d,2*h)), color=cq.Color("red"))
    
)

assy
```
[image]
[image]

adam-urbanczyk, 2021-06-22:
Also your model has [127,127,127] in FC 0.20.

jmwright, 2021-06-28:
Here is my hacked code from `occ_impl.assembly.Color` that corrects the color shift from our side, albeit with some very slight rounding error in the final STEP. Visually it's impossible to tell the difference between the original and the new STEP file though.
```python
def __init__(self, *args, **kwargs):

        if len(args) == 1:
            self.wrapped = Quantity_ColorRGBA()
            exists = Quantity_ColorRGBA.ColorFromName_s(args[0], self.wrapped)
            if not exists:
                raise ValueError(f"Unknown color name: {args[0]}")
        elif len(args) == 3:
            r, g, b = args
            self.wrapped = Quantity_Color(r, g, b ,Quantity_TOC_sRGB) #Quantity_ColorRGBA(r, g, b, 1)
            # if kwargs.get("a"):
            #     self.wrapped.SetAlpha(kwargs.get("a"))
        elif len(args) == 4:
            r, g, b, a = args
            self.wrapped = Quantity_Color(r, g, b ,Quantity_TOC_sRGB) #Quantity_ColorRGBA(r, g, b, a)
        else:
            raise ValueError(f"Unsupported arguments: {args}, {kwargs}")

    def toTuple(self) -> Tuple[float, float, float, float]:
        """
        Convert Color to RGB tuple.
        """
        # a = self.wrapped.Alpha()
        a = 1
        # rgb = self.wrapped.GetRGB()

        return (self.Red(), self.Green(), self.Blue(), a)
```
So now the question becomes, am I correcting an actual problem, or just hacking around a difference between the KiCAD viewer and OCCT's current STEP exporter?

@adam-urbanczyk I would be interested in learning more about the example you had where the STEP colors did not look right in the online viewer.

jmwright, 2021-06-29:
Here is the STEP import code for KiCAD, and it doesn't look like anything out of the ordinary is being done. https://gitlab.com/kicad/code/kicad/-/blob/master/utils/kicad2step/pcb/oce_utils.cpp#L1271

I have version 5.1 installed from the Ubuntu PPA and it lists "OpenCASCADE Community Edition: 6.9.1" as the OpenCascade version/fork that's being used. I think that is where this color mismatch is coming from. I'll add code to work around this issue that can be easily disabled later when they update their OpenCascade dependency. In their install documentation they say that OCCT >= 6.9.0 can be used during the build process, but I don't know if the newer versions of OCCT have been tried.

closed 2021-06-29 by jmwright

adam-urbanczyk, 2021-06-29:
So they are still using OCE, weird? And 6.9 is really outdated. Once they switch to OCCT 7.5 the issue might go away.

jmwright, 2021-06-29:
I was just told that they've deprecated OCE in favor of OCCT, and the Mac and Windows nightly builds use 7.5.2. I'm going to switch to running the nightly Ubuntu build to see if these problems go away.

jmwright, 2021-07-02:
The STEP color shift problem persists on KiCAD's nightly Ubuntu build, and the VRML exports do not import correctly, although they look fine in FreeCAD and the view3dscene tool in Ubuntu. I'll post more targeted issues when I have enough information.

adam-urbanczyk, 2021-07-02:
And if you use `Quantity_TOC_sRGB` then for both versions the colors are fine?

jmwright, 2021-07-02:
Yes, doing it [this way](https://gitlab.com/jmwright/kicad-packages3D-generator/-/blob/master/_tools/cq_color_correct.py#L33) seems to work fine for the STEP colors within the KiCAD viewer. FreeCAD 0.19 and 0.20 (freecad-daily) both refuse to display the colors though. I can send you the STEP file if you want to take a look.

adam-urbanczyk, 2021-07-02:
Please do send the file. It might be a FC issue.