CadQuery/cadquery #795: Generating a perforated plate

Govert123, 2021-06-23:
I would like to create a perforated plates. For each plate, I need some 10.000 holes. Basically this is very slow on my computer. 
I split the holes over a loop, and although I think that this is faster,  I still hope that someone has a pointer on how to further speed up this process.

(thickness, width, height) = (8, 1000, 1000)
(holeDistance, holeDiameter) = (15, 5)

plate = (cq.Workplane('XY')
            .box(thickness, width, height))
for j in range (1,int(height/holeDistance+1)):
    holes = [((i)*holeDistance-width/2, (j)*holeDistance-height/2) for i in range(1, int(width/holeDistance+1))]
    baffle = (baffle
                .faces(">X")
                .workplane()
                .pushPoints(holes)
                .circle(holeDiameter/2.0)
                .cutThruAll())

jmwright, 2021-06-23:
There is a similar discussion from May with an answer from @adam-urbanczyk [here](https://github.com/CadQuery/cadquery/issues/766#issuecomment-841251774).

jmwright, 2021-06-25:
@Govert123 Feel free to reopen if the above doesn't address your question.

closed 2021-06-25 by jmwright